mu-editor/mu #1453: crash report

sergeknatsko, 2021-04-09:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2021-04-09:
Hi @sergeknatsko,

We didn't really get any information with your report. What happened? Did the Mu editor crash?
If you could send us the Mu logs that would be very useful, their location should be:
- Windows: `%LOCALAPPDATA%\python\mu\Logs`
- Linus: `~/.cache/mu/`
- macOS: `~/Library/Logs/mu`

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate