mu-editor/mu #1454: CI tests segmentation fault

carlosperate, 2021-04-09:
It looks like it is happening more often. In the past 10 runs or so it seems like there is always 1 job (out of 21) that gets a seg fault:

```
tests/interface/test_main.py::test_Window_setup Fatal Python error: Segmentation fault

Thread 0x00007f141db02700 (most recent call first):
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/unittest/mock.py", line 1857 in _mock_set_magics
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/unittest/mock.py", line 1833 in __init__
  File "/home/runner/work/mu/mu/tests/interface/test_main.py", line 1531 in test_Window_setup
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/python.py", line 183 in pytest_pyfunc_call
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/python.py", line 1641 in runtest
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 162 in pytest_runtest_call
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 255 in <lambda>
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 311 in from_call
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 255 in call_runtest_hook
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 215 in call_and_report
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 126 in runtestprotocol
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/runner.py", line 109 in pytest_runtest_protocol
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/main.py", line 348 in pytest_runtestloop
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/main.py", line 323 in _main
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/main.py", line 269 in wrap_session
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/main.py", line 316 in pytest_cmdline_main
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/config/__init__.py", line 163 in main
  File "/opt/hostedtoolcache/Python/3.6.13/x64/lib/python3.6/site-packages/_pytest/config/__init__.py", line 185 in console_main
  File "/opt/hostedtoolcache/Python/3.6.13/x64/bin/pytest", line 8 in <module>
Makefile:56: recipe for target 'coverage' failed
make: *** [coverage] Segmentation fault (core dumped)
Error: Process completed with exit code 2.
```

carlosperate, 2021-04-09:
- Test Py 3.6 - ubuntu-16.04, tests/interface/test_main.py::test_Window_setup, https://github.com/mu-editor/mu/runs/2303757066?check_suite_focus=true

carlosperate, 2021-04-19:
Notes:
- It has never happened with Python 3.8 yet
- It always happens in tests/interface/test_main.py and always during a test_Window_* unit test

carlosperate, 2021-05-17:
It's been a while and a few updates since the last list, so I'll start a new one:
-  Test Py 3.6 - ubuntu-latest, tests/interface/test_main.py::test_ButtonBar_connect, https://github.com/mu-editor/mu/pull/1573/checks?check_run_id=2603240344

carlosperate, 2021-07-05:
This is still happening at the time of writing, although a lot less often:
- Test Py 3.6 - ubuntu-latest, tests/interface/test_main.py::test_Window_connect_zoom, https://github.com/mu-editor/mu/runs/2979609513?check_suite_focus=true

tonybaloney, 2022-07-19:
Try pytest-repeat with `--count=20` and run Python with `-X dev`