mu-editor/mu #1455: Pico - micropython -Empty files from device to local fs drag & drop

federicomm, 2021-04-09:
Mu 1.1.0.alpah.2 and Mu 1.1.0.beta.2 ISSUE : 
Mode: Pico
Empty file from drag & drop from device to Local FileSystem 

The copied file on Local FileSystem results to be empty ( size = 0).
Last month I have posted on YouTube a video(*) showing a workaround ( no code
changes required) and also a quickcode solution that involves small changes to 
the mu/contrib/microfs.py get() method -
(*) https://www.youtube.com/watch?v=kKziTZ-5J_4

The following are suggestions for improvements?? only,  resulting from the 
raspberrypi.org/forums feedback :

https://www.raspberrypi.org/forums/viewtopic.php?f=146&t=308117
Shift from Mu 1.02 to Mu 1.1.0.alpha.2 to work with Pi Pico board

replay from : hippy

Thanks. I now have the mu-editor working with my Pico and MicroPython though I am 
not sure it was worth the effort.
The 'file system' exposure which is what attracted me appears to have an unusable 
implementation; It's not clear where "Files on your computer" comes from, and I 
inadvertently and irrecoverably deleted files on my Pico due to the poor ergonomics 
of its design.

But I will accept others may like 'mu' more than I do and would welcome Pico and 
RP2 support.

However, as much as I appreciate hacking to make things work - and I do a lot of that
 - I don't see the virtue in hacking at the ESP port when it is trivial to create an 
 explicit Raspberry Pi Pico and RP2 port ...

mu-rp2.jpg

I would suggest someone ought to place a pull-request to do that on the official mu-editor 
build, then at least people would only have to acquire that if they want it rather than hack 
it about once they have got it. It also means it will be available for everyone as soon as 
it becomes available via the Pi package repository.
-------------------------------------------------------------------------------------------
and the following was my answer :

This quick fix makes it less easy to accidentally delete a file from the board
https://www.youtube.com/watch?v=_QTDhwhWoZk

Delete files from the device - choice by Dialog window: ( quick and temporary code solution - )

1st changes
===========
In the panes.py class MuFileList it is sufficient to clone the 
show_confirm_overwrite_dialog() method as :

show_confirm_delete_dialog()
changing msg argument, for example to
msg.setText(_( " delete ? "))
msg.setWindowTitle(_(" Are you sure?"))

or choice a more compact( or general) form
of the method with the Text and Window Title as arguments.
Code:

class MuFileList(QListWidget):
    """
    Contains shared methods for the two types of file listing used in Mu.
    """

    disable = pyqtSignal()
    list_files = pyqtSignal()
    set_message = pyqtSignal(str)

    def show_confirm_dialog(self, text, window):
        """
        Display a dialog to check if an existing file should be deleted or overwritten.
        Returns a boolean indication of the user's decision.
        """
        msg = QMessageBox(self)
        msg.setIcon(QMessageBox.Information)
        msg.setText(_(text))
        msg.setWindowTitle(_(window))
        msg.setStandardButtons(QMessageBox.Ok | QMessageBox.Cancel)
        return msg.exec_() == QMessageBox.Ok

2nd changes in:
===============
class MicropythonDeviceFileList():
in the method :
contextMenuEvent()
the quick solution is to comment the:
# if action == delete_action:
and substitute with:
if self.show_confirm_delete_dialog():

Code:
class MicroPythonDeviceFileList(MuFileList):
    """
    Represents a list of files on a MicroPython device.
    """
    put = pyqtSignal(str)
    delete = pyqtSignal(str)
    def __init__(self, home):
        super().__init__()
        self.home = home
        self.setDragDropMode(QListWidget.DragDrop)

    def dropEvent(self, event):
        source = event.source()
        text = "File " + source.currentItem().text() + " already exists on " + microfs.BOARD + " : overwrite it?"
        window = "File already exists"
        if isinstance(source, LocalFileList):
            file_exists = self.findItems(
                source.currentItem().text(), Qt.MatchExactly
            )
            if (
                not file_exists
                or file_exists
                and self.show_confirm_dialog(text, window)
            ):
                self.disable.emit()
                local_filename = os.path.join(
                    self.home, source.currentItem().text()
                )
                msg = _("Copying '{}' to " + microfs.BOARD + " .").format(local_filename)  # 2021 fm
                logger.info(msg)
                self.set_message.emit(msg)
                self.put.emit(local_filename)

    def on_put(self, microbit_file):
        """
        Fired when the put event is completed for the given filename.
        """
        msg = _("'{}' successfully copied to " + microfs.BOARD + " .").format(microbit_file) # 2021 fm
        # msg = _("'{}' successfully copied to micro:bit.").format(microbit_file) # 2021 fm
        self.set_message.emit(msg)
        self.list_files.emit()

    def contextMenuEvent(self, event):
        menu = QMenu(self)
        text = "Do you want to delete  " + self.currentItem().text() + " on " + microfs.BOARD + " ?"
        window = "DELETE !!!!"
        delete_action = menu.addAction(_("Delete (cannot be undone)"))
        action = menu.exec_(self.mapToGlobal(event.pos()))       
        if self.show_confirm_dialog(text, window): #if action == delete_action:
            self.disable.emit()
            microbit_filename = self.currentItem().text()
            logger.info("Deleting {}".format(microbit_filename))
            msg = _("Deleting '{}' from " + microfs.BOARD + " .").format(microbit_filename) # 2021 fm
            # msg = _("Deleting '{}' from micro:bit.").format(microbit_filename)
            logger.info(msg)
            self.set_message.emit(msg)
            self.delete.emit(microbit_filename)

    def on_delete(self, microbit_file):
        """
        Fired when the delete event is completed for the given filename.
        """
        msg = _("'{}' successfully deleted from " + microfs.BOARD + " .").format(
            microbit_file
        )  # 2021 fm
        # msg = _("'{}' successfully deleted from micro:bit.").format(
        #    microbit_file
        #)
        self.set_message.emit(msg)
        self.list_files.emit()

Attachments
loc_file_exist.PNG
dev_file_delete.PNG
[image]

ZanderBrown, 2021-04-11:
I'm not sure what your trying to do here but your may want to [check out markdown syntax](https://guides.github.com/features/mastering-markdown/)

federicomm, 2021-04-14:
Already solved as reference

closed 2021-04-14 by federicomm