mu-editor/mu #1456: Crash Report Mac OS Catalina 10.17.7 connected to ADAFruit 2.9" MAGTAG

ritgrad83, 2021-04-09:
[mu.log](https://github.com/mu-editor/mu/files/6286407/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Jan 12 22:13:05 PST 2021; root:xnu-6153.141.16~1/RELEASE_X86_64 x86_64

```
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/__main__.py", line 9, in <module>
    main()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/__main__.py", line 5, in main
    run()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 345, in run
    editor.setup(setup_modes(editor, editor_window))
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/logic.py", line 806, in setup
    os.makedirs(images_path)
  File "/Applications/Mu Editor.app/Contents/MacOS/../Resources/Python/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 1] Operation not permitted: '/Volumes/CIRCUITPY/images'

```

carlosperate, 2021-04-09:
Thanks for the report @ritgrad83!

It looks like it might be a permission issue in macOS. Would you be able to try these steps and check if this workaround works for you?
https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546

ritgrad83, 2021-04-09:
I did exactly as specified in #1383. I also removed MuEditor and reinstalled for good measure and still got the error. This only occurs when I have the ADAFruit MAGTAG e-ink connected running the latest version of Circuit Python 6.2.2. When I disconnect from the device - Mu Editor starts up but with an error that no Circuit Python devices were connected

Also attached a log file

[mu.log](https://github.com/mu-editor/mu/files/6286862/mu.log)

[image]

ritgrad83, 2021-04-09:
Another update - I restarted Mu-editor without the MAG TAG connected and bypassed the "Could not find CP device" error so that it uses my computer's local drive to find files. Then I reconnected the MAG TAG and it was able to find the device. 

I tried again - starting MuEditor with MAG TAG device already connected and still got the system crash error. 

When I disconnected the MAG TAG device and then restarted MuEditor **without** the device connected - repeating the first step - bypassing the "Could not find CP device" error and then connected the device - it connects.

So in summary - start Mu-Editor first and then connect to prevent system crash - not a graceful workaround

carlosperate, 2021-10-06:
Hi @ritgrad83,

This should have been fixed in beta.6: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.6

More info in https://github.com/mu-editor/mu/issues/1383#issuecomment-935760258.

Could you try this version to see if it works?

carlosperate, 2021-12-27:
We believe this has been fixed in beta.6, so I'll close this as resolved. Thanks for the report!

closed 2021-12-27 by carlosperate