mu-editor/mu #1457: Windows: The venv verifying modules step launches command prompt windows

carlosperate, 2021-04-09:
When Mu verifies the modules in the venv it opens background command prompt windows, which can be a bit "scary looking". 

I haven't experienced this in macOS nor Linux, so it looks like a Windows-only thing.

Would it be possible to hide these?
Just a guess, but maybe it's possible the modules are checked with pythonw.exe and we could avoid the extra windows with python.exe?

https://user-images.githubusercontent.com/4189262/114197719-856bdb80-994a-11eb-9ba1-3c82b448146a.mp4

tjguk, 2021-04-09:
@carlosperate  that's very odd. As you might imagine, I've run this process a few times myself on Windows, and I've *never* seen it launch background console windows. I note that you're running Win8.1 which *might* have something do to with it -- but I'm surprised.

Running `pythonw.exe` is probably possible, but it tends to interfere with stdout|err channels, so there might be complications.

carlosperate, 2021-04-09:
I'm seeing the same thing in Windows 10:

https://user-images.githubusercontent.com/4189262/114200707-5efb6f80-994d-11eb-95c4-9f5a718db16c.mp4

tjguk, 2021-04-10:
@carlosperate is that running from the installer? Or from a dev venv? (Just so I can reproduce...)

carlosperate, 2021-04-10:
In both cases it was from the installer.

carlosperate, 2021-08-10:
I assume this is because when running from the installer Mu is launched via `pythonw.exe`, and so the `virtual_environment.py` module launches a new process with `sys.executable`, which also uses `pythonw.exe` and we see all the terminal windows open and close.

We can update `virtual_environment.py` to:
- Check if the `sys.executable` ends in  `w.exe`,
    - If it does, check if there is a version of the exe without the `w` (so `python.exe` instead of `pythonw.exe`)
        - If it does, launch with that executable instead

closed 2021-12-20 by ntoll