mu-editor/mu #1463: Un problème est survenu pendant le flashage de la micro:bit.

profroques, 2021-04-09:
Bonjour,
je suis sur Mac Catalina. J'ai installé la dernière version de Mu-editor (Mu.Editor.1.1.0b3) sans problème. Lorsque je flash la carte micro:bit, j'ai ce message d'erreur qui apparait : 
Un problème est survenu pendant le flashage de la micro:bit. Ne pas déconnecter le périphériques tant que le flashage n'est pas terminé. Regarde les logs pour plus d'informations.

Auriez vous déjà été confrontés à ce problème ?

Merci pour votre aide :)

tjguk, 2021-04-10:
@profroques  -- would you be able to provide the logs from Mu onto this ticket, please? Ideally as an attached file

There are instructions here:

https://codewith.mu/en/howto/1.1/read_logs

carlosperate, 2021-08-10:
Hi @profroques,

Are you still experiencing this issue? if you are, could you update to the latest beta version, try again and if the problem persists send the log? Should be located in `~/Library/Logs/mu`

carlosperate, 2021-09-21:
Hi @profroques 
If you are in macOS and having issues flashing some board could you try the steps listed here? https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546

carlosperate, 2021-10-06:
Hi @profroques,

This should have been fixed in beta.6: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.6

More info in https://github.com/mu-editor/mu/issues/1383#issuecomment-935760258.

closed 2021-10-06 by carlosperate