mu-editor/mu #1466: Mu 1.1.0-beta-3 Unable to start after installation

Monster-DP, 2021-04-10:
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[image]

tjguk, 2021-04-10:
@Monster-DP what platform are you on? (I'm guessing it's Windows because of the reference to pythonw).

Can you see if there are any logs in?

`%LOCALAPPDATA%\python\mu\logs`

It's quite possible that there aren't, but if there are could upload here please?

Monster-DP, 2021-04-10:
2021-04-10 16:46:38,816 - root:266(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.3
2021-04-10 16:46:38,816 - root:267(run) INFO: uname_result(system='Windows', node='LAPTOP-84ECJJA5', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 140 Stepping 1, GenuineIntel')
2021-04-10 16:46:38,816 - root:268(run) INFO: Python path: ['C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-04-10 16:46:38,827 - root:269(run) INFO: Language code: zh_CN
2021-04-10 16:47:06,882 - root:266(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.3
2021-04-10 16:47:06,882 - root:267(run) INFO: uname_result(system='Windows', node='LAPTOP-84ECJJA5', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 140 Stepping 1, GenuineIntel')
2021-04-10 16:47:06,882 - root:268(run) INFO: Python path: ['C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\木羊羽\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-04-10 16:47:06,882 - root:269(run) INFO: Language code: zh_CN

Monster-DP, 2021-04-10:
[mu.log](https://github.com/mu-editor/mu/files/6289526/mu.log)

Monster-DP, 2021-04-10:
> @Monster-DP what platform are you on? (I'm guessing it's Windows because of the reference to pythonw).
> 
> Can you see if there are any logs in?
> 
> `%LOCALAPPDATA%\python\mu\logs`
> 
> It's quite possible that there aren't, but if there are could upload here please?

[mu.log](https://github.com/mu-editor/mu/files/6289529/mu.log)

tjguk, 2021-04-10:
Thanks @Monster-DP  -- there's not an obvious cause; I suspect it's a Qt configuration issue, but we'll have to search around a bit to see what the error means. (The error is coming from Qt, the UI framework we use, not from our code).

carlosperate, 2021-04-10:
I think I recall seeing this error message years ago when we were using PyInstaller and some dlls were not correctly packaged.
@Monster-DP have you tried uninstalling the application and installing it again?

carlosperate, 2021-08-10:
Hi @Monster-DP, I was wondering if you were able to uninstall Mu, install the latest beta and try again?

carlosperate, 2021-12-17:
Hi @Monster-DP, I think this crash might be related to issue: #1831

We think we might have found a solution to that problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.