mu-editor/mu #1471: Crash Report

bnystrom, 2021-04-12:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\bnystrom\AppData\Local\python\mu\mu_venv-38-20210412-001122 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 583, in create
    subprocess.run(
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\bnystrom\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\bnystrom\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\bnystrom\\AppData\\Local\\python\\mu\\mu_venv-38-20210412-001547']' returned non-zero exit status 1.

```

carlosperate, 2021-04-12:
Thanks for the report @bnystrom!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

bnystrom, 2021-04-13:
[mu.log](https://github.com/mu-editor/mu/files/6301074/mu.log)

bnystrom, 2021-04-13:
Strange that although I fully removed the 1.1.0-alpha, this is the string it gives in the latest log file. I actually used "Add-Remove" apps (Windows 10) to remove both versions, and then reinstalled the latest. The crash report was from after removing both and adding back just the beta. 

My system had no issues running the alpha release.

Hope the log file helps. Let me know if you need anything else.

carlosperate, 2021-04-13:
Thanks @bnystrom!

The first entry in the log is for the alpha, the ones below belong to the beta.3, so that's okay.

Could you run this command in a command prompt (you can open one by typing cmd.exe in the Windows start menu) and copy/paste the output here?

```
"C:\\Users\\bnystrom\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe" -m virtualenv -p "C:\\Users\\bnystrom\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe" "C:\\Users\\bnystrom\\AppData\\Local\\python\\mu\\mu_venv-test"
```

It will likely create a folder `C:\\Users\\bnystrom\\AppData\\Local\\python\\mu\\mu_venv-test`, which you can delete after the test.

bnystrom, 2021-04-15:
Sorry for the late reply.

I executed the command line above and it produced nothing. No console
output and no mu_venv-test file.

Best Regards
Bryan Nystrom

carlosperate, 2021-04-20:
Thanks for testing again @bnystrom, I think maybe I should have remove the double slashes (`\\`) in the command, could you try this one?

```
"C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" -m virtualenv -p "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" "C:\Users\bnystrom\AppData\Local\python\mu\mu_venv-test"
```

bnystrom, 2021-05-06:
Sorry again for another late reply. Bad news and good news.

Bad news is I wasn't able to get what you asked for to debug the problem
because... the good news is I uninstall beta 3 and reinstall beta 4 and now
it seems to be loading and working fine.

Best Regards
Bryan Nystrom

tjguk, 2021-05-06:
Thanks for the update, @bnystrom -- I'm glad that our fixes for Beta 4 worked for someone  :)

closed 2021-05-06 by tjguk