mu-editor/mu #1472: venv subprocesses do not log their stdout/stderr when the exit is non-zero

carlosperate, 2021-04-13:
For example this case from https://github.com/mu-editor/mu/issues/1471, but there is many more examples in other crash reports:

```
2021-04-12 00:15:47,708 - mu.virtual_environment:579(create) INFO: Creating virtualenv: C:\Users\bnystrom\AppData\Local\python\mu\mu_venv-38-20210412-001547
2021-04-12 00:15:47,709 - mu.virtual_environment:580(create) INFO: Virtualenv name: mu_venv-38-20210412-001547
2021-04-12 00:15:54,871 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\bnystrom\AppData\Local\python\mu\mu_venv-38-20210412-001122 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 583, in create
    subprocess.run(
  File "C:\Users\bnystrom\AppData\Local\Programs\Mu Editor\Python\lib\subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\bnystrom\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\bnystrom\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\bnystrom\\AppData\\Local\\python\\mu\\mu_venv-38-20210412-001547']' returned non-zero exit status 1.
```

We should look into being able to print the stdout/stderr into the log as well, as that will help the debugging of crash reports.

carlosperate, 2021-05-12:
In a similar vein we have cases like this one, where the `pip install` seems to end early and no error message is captured in the log:

> [mu.log](https://github.com/mu-editor/mu/files/6464533/mu.log)
> Version: 1.1.0.beta.4
> Locale: en_GB
> Platform: Windows 10 10.0.18362 AMD64

Looks like it fails to install some of the packaged wheels, but it doesn't throw/log an error:

```
2021-05-12 08:36:36,954 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-05-12 08:39:39,846 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ecdsa-0.16.1-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\esptool-3.0-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\flask-1.1.2-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipykernel-5.5.3-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython-7.22.0-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython_genutils-0.2.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\itsdangerous-1.1.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jedi-0.18.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jinja2-2.11.3-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_client-6.1.12-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jupyter_core-4.7.1-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\markupsafe-1.1.1-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\numpy-1.20.2-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\parso-0.8.2-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pgzero-1.2.1-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pickleshare-0.7.5-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\prompt_toolkit-3.0.18-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pycparser-2.20-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pygame-2.0.1-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pygments-2.8.1-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pyserial-3.5-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\python_dateutil-2.8.1-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pywin32-300-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\pyzmq-22.0.3-cp38-cp38-win_amd64.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\qtconsole-4.7.4-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\qtpy-1.9.0-py2.py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\reedsolo-1.5.4-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\setuptools-56.0.0-py3-none-any.whl
Processing c:\users\alice\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\six-1.15.0-py2.py3-none-any.whl
[... cut ...]
Installing collected packages: ipython-genutils, wcwidth, traitlets, six, pywin32, parso, tornado, setuptools, pyzmq, python-dateutil, Pygments, pycparser, prompt-toolkit, pickleshare, jupyter-core, jedi, decorator, colorama, backcall, MarkupSafe, jupyter-client, ipython, cffi, Werkzeug, reedsolo, QtPy, pyserial, pygame, numpy, Jinja2, itsdangerous, ipykernel, ecdsa, cryptography, click, bitstring, qtconsole, pgzero, Flask, esptool, adafruit-board-toolkit

  Attempting uninstall: setuptools
    Found existing installation: setuptools 54.1.2
    Uninstalling setuptools-54.1.2:
      Successfully uninstalled setuptools-54.1.2
```

tjguk, 2021-05-17:
I really can't see anything obvious in the code. In case it helps, I've changed the code to explictly pipe stdout and stderr independently and combined them later

I wonder whether pip doesn't actually use stdout/err for some of it's output. I'll try a file redirection and see what happens...

tjguk, 2021-05-17:
[pip.stderr.log](https://github.com/mu-editor/mu/files/6496463/pip.stderr.log)
[pip.stdout.log](https://github.com/mu-editor/mu/files/6496464/pip.stdout.log)

By way of reference, this is the pip output from doing `pip install --upgrade flask`

tjguk, 2021-05-17:
[pip.stderr.log](https://github.com/mu-editor/mu/files/6496478/pip.stderr.log)

And this is stderr if I try to install a package pip can't find. (stdout is empty)

So by way of a guess, it doesn't look as though we're picking up stderr from the pip processes. (Assuming that there's some kind of error output which is preventing the packages being installed)

carlosperate, 2021-08-10:
We can close this as resolved.

closed 2021-08-10 by carlosperate