mu-editor/mu #1473: Windows installer: Add option to install for all users

carlosperate, 2021-04-13:
Right now Mu is installed in a user-specific location and not available to all users.
Previous Mu installers had the option to be installed globally (in program files I believe?), it would be nice to have this option via pup as well.

carlosperate, 2021-04-15:
Looks like it might also be important in school networks: https://github.com/mu-editor/mu/issues/1477

kassn-bms, 2021-04-15:
Yes indeed it would be good to have an option of where it installs - or perhaps even if you don't want to change the msi installer, providing an MST file with it that specifies a different location such as program files would be extremely helpful!

carlosperate, 2021-04-15:
Thanks for all the help @kassn-bms!
I wasn't aware of [MST files](https://www.exemsi.com/documentation/mst-transformation-files/), something to look into. Some links I've found that might be relevant:
- https://docs.microsoft.com/en-us/archive/blogs/alexshev/from-msi-to-wix-part-7-customizing-installation-using-transforms
- https://docs.microsoft.com/en-gb/windows/win32/msi/database-transforms

I assume it might be easier to offer the option during installation, would that be good enough for this kind of set up?

kassn-bms, 2021-04-15:
For our set up, and this might be the case for a number of other schools also, having the option come up during the installation might not be a good idea as we install MSI's through group policy for automation, and that happens when the computer is starting up and is applying the policies so the software is installed **before** it gets to the login screen.

That's why I think offering an MST file with the MSI would be the best option for schools, as we can then edit the MST with something called "InstEd It!" ([here is a link to the download if it helps at all](http://www.instedit.com/)) ourselves to specify where we want it installed then we just add the MSI with the MST file into the software installation policy, and that way you can still keep the MSI to install on a per-user basis as a default.

Hope this helps!

carlosperate, 2022-10-07:
Current instructions we can put in the Mu docs: https://pup.readthedocs.io/en/latest/platform_notes/windows_msi.html#per-user-vs-per-machine-installs