mu-editor/mu #1474: Crash Report

KellyPared, 2021-04-13:
[Mu Crash Report.pdf](https://github.com/mu-editor/mu/files/6306162/Mu.Crash.Report.pdf)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.2
Locale: en_US
Platform: Darwin 18.7.0 Darwin Kernel Version 18.7.0: Tue Nov 10 00:07:31 PST 2020; root:xnu-4903.278.51~1/RELEASE_X86_64 x86_64

```
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/interface/main.py", line 254, in on_modified
    state_lbl.setPixmap(load_pixmap("document-dirty.svg"))
RuntimeError: wrapped C/C++ object of type QLabel has been deleted

```
Traceback (most recent call last):
File "/Applications/Mu
Editor.app/Contents/Resources/Python/lib/python3.8/sitepackages/
mu/interface/main.py", line
254, in on_modified
state_lbl.setPixmap(load_pixmap("documentdirty.
svg"))
RuntimeError: wrapped C/C++ object of
type QLabel has been deleted

carlosperate, 2021-04-14:
Thank you for the report @KellyPared!
Would you be able to also attach the Mu log to this GitHub issue? That will really help us with the debugging. File should be located in: `~/Library/Logs/mu`

Do you remember what were you doing in the Mu editor when it crashed? Did you open/closed a new tab or something or something like that?

KellyPared, 2021-04-14:
I was working with the Microbit and US sensor. I always have tons of tabs open because I teach multiple sections. and then Mu shut down.

[
[mu.log.2021-04-13.txt](https://github.com/mu-editor/mu/files/6310178/mu.log.2021-04-13.txt)

]

carlosperate, 2021-04-14:
Thanks for the log @KellyPared!

Looking at the events there wasn't much logged right before the crash:
```
2021-04-13 15:07:09,319 - mu.logic:962(new) INFO: Added a new tab.
2021-04-13 15:15:29,257 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/interface/main.py", line 254, in on_modified
    state_lbl.setPixmap(load_pixmap("document-dirty.svg"))
RuntimeError: wrapped C/C++ object of type QLabel has been deleted
```

A new tab was created several minutes before the issue, which happened in this event handler (`state_lbl` is a `QLabel`):
https://github.com/mu-editor/mu/blob/6359ac12189065118ab9467a5847151ce7894dbb/mu/interface/main.py#L248-L257

We should check the tab or it's labels are still "alive" before performing actions on them.

@ZanderBrown I believe this feature was by you a couple of years ago, any thoughts as to what might be causing this? Is it a race condition or maybe something else?

ntoll, 2021-04-14:
Looking at the code, it appears the `on_modified` handler has been called after the `widget` object has been GC'd. Perhaps it was in the queue for the next iteration of the event loop **after** the widget was destroyed. If there's a chance that can happen, then we should definitely check before calling `setPixmap`. :rocket: :boom: :scream: :wink:

carlosperate, 2021-04-14:
Yeah, I was thinking the same, as far as I can see the label should only be destroyed if the tab is closed, and trying to close an unsaved tab opens a confirmation window that the user has to accept/cancel.
All these steps are triggered by the user, which means at least several hundreds of ms between each step (editing the tab content, clicking the close button, accepting/cancelling the prompt), it's kind of hard to see how the event queue could be so far behind, it's a bit weird.

ntoll, 2021-07-10:
PR with fix is #1694.

PR #1694 (2021-07-19): Wrap QLabel setPixmap call in event handler in a try/except to avoid RuntimeError exceptions.
Fixes #1474.

Wraps code that references a QLabel that may be deleted by Qt before parent Python object is GC'd. Comments in code explain why I've used the `try/except` approach.
closed 2021-07-19 by ntoll