mu-editor/mu #1475: Crash Report

Robert021965, 2021-04-14:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Linux 5.4.88-12233-g8e0826df26a0 #1 SMP PREEMPT Wed Apr 7 19:44:59 PDT 2021 x86_64

```
Traceback (most recent call last):
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/robertogiusto/.local/share/mu/mu_venv-37-20210414-083748 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 115, in download
    check=True,
  File "/usr/lib/python3.7/subprocess.py", line 487, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'wheel', '--no-deps', '--wheel-dir', '/home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels', '/home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels/cryptography-3.4.7.tar.gz']' returned non-zero exit status 1.

```

Robert021965, 2021-04-14:
New to Linux, new to Python 🤦

Using Linux on a Chromebook to run Python

carlosperate, 2021-04-14:
Thanks for the repot @Robert021965!

Could you run the following commands on a terminal and copy/paste the output here? It will help us with the debugging:

```
/usr/bin/python3 -m pip --version
```

```
/usr/bin/python3 -m pip wheel --no-deps --wheel-dir /home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels /home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels/cryptography-3.4.7.tar.gz
```

Robert021965, 2021-04-14:
>
>
> Could you run the following commands on a terminal and copy/paste the
> output here? It will help us with the debugging:
>
> /usr/bin/python3 -m pip --version
>
> pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)

> /usr/bin/python3 -m pip wheel --no-deps --wheel-dir /home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels /home/robertogiusto/.local/lib/python3.7/site-packages/mu/wheels/cryptography-3.4.7.tar.gz
 Processing
./.local/lib/python3.7/site-packages/mu/wheels/cryptography-3.4.7.tar.gz
  Installing build dependencies ... done
Building wheels for collected packages: cryptography
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools,
tokenize;__file__='/tmp/pip-req-build-8nd79xwq/setup.py';f=getattr(tokenize,
'open', open)(__file__);code=f.read().replace('\r\n',
'\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d
/tmp/pip-wheel-xojq4yfo:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.7
  creating build/lib.linux-x86_64-3.7/cryptography
  copying src/cryptography/__about__.py ->
build/lib.linux-x86_64-3.7/cryptography
  copying src/cryptography/__init__.py ->
build/lib.linux-x86_64-3.7/cryptography
  copying src/cryptography/exceptions.py ->
build/lib.linux-x86_64-3.7/cryptography
  copying src/cryptography/fernet.py ->
build/lib.linux-x86_64-3.7/cryptography
  copying src/cryptography/utils.py ->
build/lib.linux-x86_64-3.7/cryptography
  creating build/lib.linux-x86_64-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/__init__.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/_der.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/_oid.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat
  copying src/cryptography/hazmat/_types.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat
  creating build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/__init__.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/base.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/certificate_transparency.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/extensions.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/general_name.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/name.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/ocsp.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  copying src/cryptography/x509/oid.py ->
build/lib.linux-x86_64-3.7/cryptography/x509
  creating build/lib.linux-x86_64-3.7/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/__init__.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat/backends
  copying src/cryptography/hazmat/backends/interfaces.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat/backends
  creating build/lib.linux-x86_64-3.7/cryptography/hazmat/bindings
  copying src/cryptography/hazmat/bindings/__init__.py ->
build/lib.linux-x86_64-3.7/cryptography/hazmat/bindings
[... 259 lines cut ...]
  error: can't find Rust compiler

  If you are using an outdated pip version, it is possible a prebuilt wheel
is available for this package but pip is not able to install from it.
Installing from the wheel would avoid the need for a Rust compiler.

  To update pip, run:

      pip install --upgrade pip

  and then retry package installation.

  If you did intend to build this package from source, try installing a
Rust compiler from your system package manager and ensure it is on the PATH
during installation. Alternatively, rustup (available at https://rustup.rs)
is the recommended way to download and update the Rust compiler toolchain.

  This package requires Rust >=1.41.0.

  ----------------------------------------
  Failed cleaning build dir for cryptography
Failed to build cryptography
ERROR: Failed to build one or more wheels

Thanks for your help

carlosperate, 2021-04-14:
Thanks @Robert021965, that more or less confirms my suspicion that perhaps the pip version is too old.

Did you try to run `/usr/bin/python3 -m pip --version` as well? What version of pip does it report?

If you can update pip with this command it should fix the issue:

```
/usr/bin/python3 -m pip install --upgrade pip
```

Robert021965, 2021-04-14:
It worked! Thank you so much!

carlosperate, 2021-04-14:
I'm glad you were able to resolve it, thanks for all the testing!

closed 2021-04-14 by carlosperate