mu-editor/mu #1476: Crash Report

neilmlynn, 2021-04-14:
[mu.log](https://github.com/mu-editor/mu/files/6311142/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\User\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\User\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\User\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

neilmlynn, 2021-04-14:
Tried to install Mu-editor twice; failed both times.

Tried to install Mu-editor on my Raspberry Pi 4 twice, using their Buster system or with  pip .  Failed.

carlosperate, 2021-04-20:
Thanks for the report @neilmlynn. I'll reopen this issue as I assume you haven't been able to run Mu yet?

Could you run this command in a command prompt in your Windows computer (you can open one by typing `cmd.exe` in the Windows start menu) and copy/paste the output to this issue? It'll be really helpful for us:

```
"C:\Users\User\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" -m virtualenv -p "C:\Users\User\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" "C:\Users\User\AppData\Local\python\mu\mu_venv-test"
```

It will likely create a folder `C:\Users\User\AppData\Local\python\mu\mu_venv-test`, which you can delete after the test.

neilmlynn, 2021-04-20:
Carlos,

Many thanks for your reply.   Since sending off the help message, I have been unable to go back to the problem.  Retirement is another description for ‘no free time’.

So yesterday, I went back to my laptop (HP, Windows 10).   I erased the previous installation of Mu (which was my second try at installation), and reinstalled it.   That’s all I did.   Amazing, it ran first time.

My Raspberry Pi model 4 had similar problems and I tried 3 times to erase and re-install, using Command line or GUI interface.   On the Web, someone suggested that Mu installation could be conflicting with other previously installed programs so I started the Raspberry Pi installation from a clean SD card and a fresh operating system.
Then Mu installed and ran.

So, all OK.    I will keep your notes, just in case.

Neil
1519
20/04/21

carlosperate, 2021-04-20:
Thanks for testing again, glad you got it working!
I'll close this for now, but please feel free to reopen this issue, or open a new one, if you encounter any issues.

Thanks again!

closed 2021-04-20 by carlosperate