mu-editor/mu #1477: Crash Report - Unable to find log file (location not known)

kassn-bms, 2021-04-14:
**REMEMBER TO ATTACH YOUR LOG FILE** - Unable to find log file (location not known)

I know this is a group policy issue, however I need to know what the subprocess.py is trying to start / open at that point.

I've created a hash rule which un-restricts that file, but there is obviously something else that it's trying to start which we are currently still blocking.

Any help / info would be much appreciated!

Version: 1.1.0.beta.2
Locale: en_GB
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 441, in ensure_and_create
    self.ensure()
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 469, in ensure
    self.ensure_interpreter_version()
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 522, in ensure_interpreter_version
    process = subprocess.run(
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "\\bushey-meads.internal\packages\Applications\Python MU Editor\Mu Editor\Python\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 1260] This program is blocked by group policy. For more information, contact your system administrator

```

kassn-bms, 2021-04-14:
here is the link to the error (hoping this will work)

[Error Code Link](https://codewith.mu/crash/?v=1.1.0.beta.2&l=en_GB&p=V2luZG93cyAxMCAxMC4wLjE4MzYyIEFNRDY0&e=VHJhY2ViYWNrIChtb3N0IHJlY2VudCBjYWxsIGxhc3QpOgogIEZpbGUgIlxcYnVzaGV5LW1lYWRzLmludGVybmFsXHBhY2thZ2VzXEFwcGxpY2F0aW9uc1xQeXRob24gTVUgRWRpdG9yXE11IEVkaXRvclxQeXRob25cbGliXHNpdGUtcGFja2FnZXNcbXVcYXBwLnB5IiwgbGluZSAxNjIsIGluIHJ1bgogICAgcmFpc2UgZXgKICBGaWxlICJcXGJ1c2hleS1tZWFkcy5pbnRlcm5hbFxwYWNrYWdlc1xBcHBsaWNhdGlvbnNcUHl0aG9uIE1VIEVkaXRvclxNdSBFZGl0b3JcUHl0aG9uXGxpYlxzaXRlLXBhY2thZ2VzXG11XGFwcC5weSIsIGxpbmUgMTQ5LCBpbiBydW4KICAgIHZlbnYuZW5zdXJlX2FuZF9jcmVhdGUoc2VsZi5kaXNwbGF5X3RleHQpCiAgRmlsZSAiXFxidXNoZXktbWVhZHMuaW50ZXJuYWxccGFja2FnZXNcQXBwbGljYXRpb25zXFB5dGhvbiBNVSBFZGl0b3JcTXUgRWRpdG9yXFB5dGhvblxsaWJcc2l0ZS1wYWNrYWdlc1xtdVx2aXJ0dWFsX2Vudmlyb25tZW50LnB5IiwgbGluZSA0NDEsIGluIGVuc3VyZV9hbmRfY3JlYXRlCiAgICBzZWxmLmVuc3VyZSgpCiAgRmlsZSAiXFxidXNoZXktbWVhZHMuaW50ZXJuYWxccGFja2FnZXNcQXBwbGljYXRpb25zXFB5dGhvbiBNVSBFZGl0b3JcTXUgRWRpdG9yXFB5dGhvblxsaWJcc2l0ZS1wYWNrYWdlc1xtdVx2aXJ0dWFsX2Vudmlyb25tZW50LnB5IiwgbGluZSA0NjksIGluIGVuc3VyZQogICAgc2VsZi5lbnN1cmVfaW50ZXJwcmV0ZXJfdmVyc2lvbigpCiAgRmlsZSAiXFxidXNoZXktbWVhZHMuaW50ZXJuYWxccGFja2FnZXNcQXBwbGljYXRpb25zXFB5dGhvbiBNVSBFZGl0b3JcTXUgRWRpdG9yXFB5dGhvblxsaWJcc2l0ZS1wYWNrYWdlc1xtdVx2aXJ0dWFsX2Vudmlyb25tZW50LnB5IiwgbGluZSA1MjIsIGluIGVuc3VyZV9pbnRlcnByZXRlcl92ZXJzaW9uCiAgICBwcm9jZXNzID0gc3VicHJvY2Vzcy5ydW4oCiAgRmlsZSAiXFxidXNoZXktbWVhZHMuaW50ZXJuYWxccGFja2FnZXNcQXBwbGljYXRpb25zXFB5dGhvbiBNVSBFZGl0b3JcTXUgRWRpdG9yXFB5dGhvblxsaWJcc3VicHJvY2Vzcy5weSIsIGxpbmUgNDg5LCBpbiBydW4KICAgIHdpdGggUG9wZW4oKnBvcGVuYXJncywgKiprd2FyZ3MpIGFzIHByb2Nlc3M6CiAgRmlsZSAiXFxidXNoZXktbWVhZHMuaW50ZXJuYWxccGFja2FnZXNcQXBwbGljYXRpb25zXFB5dGhvbiBNVSBFZGl0b3JcTXUgRWRpdG9yXFB5dGhvblxsaWJcc3VicHJvY2Vzcy5weSIsIGxpbmUgODU0LCBpbiBfX2luaXRfXwogICAgc2VsZi5fZXhlY3V0ZV9jaGlsZChhcmdzLCBleGVjdXRhYmxlLCBwcmVleGVjX2ZuLCBjbG9zZV9mZHMsCiAgRmlsZSAiXFxidXNoZXktbWVhZHMuaW50ZXJuYWxccGFja2FnZXNcQXBwbGljYXRpb25zXFB5dGhvbiBNVSBFZGl0b3JcTXUgRWRpdG9yXFB5dGhvblxsaWJcc3VicHJvY2Vzcy5weSIsIGxpbmUgMTMwNywgaW4gX2V4ZWN1dGVfY2hpbGQKICAgIGhwLCBodCwgcGlkLCB0aWQgPSBfd2luYXBpLkNyZWF0ZVByb2Nlc3MoZXhlY3V0YWJsZSwgYXJncywKT1NFcnJvcjogW1dpbkVycm9yIDEyNjBdIFRoaXMgcHJvZ3JhbSBpcyBibG9ja2VkIGJ5IGdyb3VwIHBvbGljeS4gRm9yIG1vcmUgaW5mb3JtYXRpb24sIGNvbnRhY3QgeW91ciBzeXN0ZW0gYWRtaW5pc3RyYXRvcgo%3D)

tjguk, 2021-04-14:
Thanks @kassn-bms  -- it's this code here:

https://github.com/mu-editor/mu/blob/6359ac12189065118ab9467a5847151ce7894dbb/mu/virtual_environment.py#L511

It's running the Python interpreter which which will be used for Mu to run user code. (And checking that it's running the same version as the interpreter actually running Mu)

The command is [slightly simplifed] `%LOCALAPPDATA%\python\mu\mu_venv_[unique timestamp]\scripts\python.exe -c "import sys; print(sys.version)`

tjguk, 2021-04-14:
NB I'm guessing you're working at a school. We're very keen to make Mu work for school admins, teachers etc., so please let us know what we might do to remove any pain points from your point of view.

tjguk, 2021-04-14:
BTW it's interesting that we can't detect your log location. In general on Windows it would be in `%LOCALAPPDATA%\python\mu\logs` but it's possible that Group pols and or shell redirection has moved that location.

kassn-bms, 2021-04-14:
Hi tjguk,

Thank you for your swift response!!

Yes we're hoping to roll Mu out across our entire school to allow all kids access to it as it's great!
I noticed that upon installing Mu (not sure if this is the case everywhere or just on our, I'll be honest, weirdly set up network) it seems to install it in C:\Users\%username%\AppData\Local\Programs\Mu Editor - which for obvious reasons we can't allow students to access so that's quite a big issue for us.

I've managed to get around it by placing the Mu Editor folder into one of our network shares that students can run the vbs script from, at which point we get this error.

Are you able to clarify for me how I might be able to unblock and allow the interpreter to run?

My suggestion would be to make Mu install in Program Files, or at least an option to change where it installs to, as most schools as far as I know allow students to execute files, software etc. from there which would make it fairly simple to install across the network with the MSI then just having a shortcut to the VBS script to run Mu.

Hope this helps, kindly let me know regarding the interpreter when you can!

Thanks!

tjguk, 2021-04-14:
I'm actually a little surprised that you restrict access to the user's own Local AppData area -- isn't that where user-installed programs are supposed to be? Certainly Python itself ends up there for a per-user installation and a couple of other random things I have installed. Or did I misunderstand your point?

I suspect it would be worth our having a call about this, so I can understand where the issues are your side, and perhaps I can explain what the editor is trying to do when you start up which is hitting up against your GPOs. If you'd be willing, then my email is on my Github profile: https://github.com/tjguk/  -- drop me a line and we can arrange something.

kassn-bms, 2021-04-15:
Yeah it's because everyone here uses Roaming profiles instead of local so anywhere and everywhere they login they have access to the same folders etc. - though you're right that's where they're supposed to be, but we don't use a lot of user-installed programs as all the software we have are installed from a central network share through group policy and generally most things install into Program Files from where everyone can read & execute from.

For python the MSI installs with an MST that tells it to go into C:\Python33\ instead and everyone can run it from there.

That would be brilliant if you have some time to look at this with me, the actual running of Mu I can get around by placing it into the central network share then having a shortcut in the start menu to run the VBS which starts Mu, just need to figure out which policy is blocking the interpreter from running.

I will send you an email shortly - thank you!

kassn-bms, 2021-04-15:
@carlosperate Thank you!