mu-editor/mu #1478: Crash Report

EduardoDestefani, 2021-04-15:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: pt_BR
Platform: Linux 5.8.0-49-generic #55~20.04.1-Ubuntu SMP Fri Mar 26 01:01:07 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/eduardo/anaconda3/lib/python3.8/site-packages/mu/app.py", line 162, in run
    raise ex
  File "/home/eduardo/anaconda3/lib/python3.8/site-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/home/eduardo/anaconda3/lib/python3.8/site-packages/mu/virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```

carlosperate, 2021-04-15:
Thanks for the report @EduardoDestefani!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
Thanks for sending the crash report!

closed 2021-08-10 by carlosperate