mu-editor/mu #1479: Crash Report

tonygo, 2021-04-16:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\tonyg\AppData\Local\python\mu\mu_venv-38-20210416-115214 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 299, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 598, in create
    self.register_baseline_packages()
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\tonyg\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 305, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

tonygo, 2021-04-16:
Windows just been updated on but my laptop and Pc. Installation took ages and then fell over

carlosperate, 2021-04-16:
Thanks for the report @tonygo!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

tonygo, 2021-04-16:
sorry nothing there. I've had uninstall all of mu and put an older version back in.regardsTonySent from my Galaxy

carlosperate, 2021-04-16:
Are you sure you don't have any files in `C:\Users\tonyg\AppData\Local\python\mu\Logs`? even after uninstalling and reinstalling an old version there would likely be a log file or two left there.

tonygo, 2021-04-16:
I've got 1.1.0-beta3 working on both my windows 10 machines now. 

On the previous attempts the installation died with a parsing error - while the big snake was on the screen. 

I think the previous attempts died because I had not uninstalled the old versions - 1.0.2

This time I made sure that all the old versions were removed before trying to install the new version.

Might be useful to tell users to uninstall all old versions first.

Sorry to trouble you

Regards

Tony

(PS I love your Mu editor!)

carlosperate, 2021-04-16:
Hi @tonygo,

No trouble at all, on the contrary, we really appreciate the feedback and bug reports, they are really valuable to us!

In theory having the old version should not affect installation of the new betas, which is why we are quite interested to debug this issue and have a look at the Mu logs, as it might contain some extra information about why it failed.
As you've installed and uninstalled a couple of versions it would be really useful if you could check if there are still any log files in `C:\Users\tonyg\AppData\Local\python\mu\Logs` and post them all to this GitHub issue?
If you are responding to this via email adding attachments won't work (in case you've tried to send the files in your previous emails), but you can visit this issue in the browser and drag&drop the files into a new response: https://github.com/mu-editor/mu/issues/1479

Thanks again!

tonygo, 2021-04-16:
I'm really sorry but I have searched my drive and cannot find any .\mu\Logs

I cannot even find a tonyg folder in Users on drive C:

Regards

Tony Goodhew

carlosperate, 2021-04-19:
Thanks @tonygo, well, you got it working in the end, which is what matters!
I'll close this issue for now, but if you experience this issue again it'd be great if you could reopen it letting us know.

Thank you again for all the testing!

closed 2021-04-19 by carlosperate