mu-editor/mu #1481: Crash Report

ldale15, 2021-04-17:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\ldale\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\ldale\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\ldale\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.

```
[mu.log](https://github.com/mu-editor/mu/files/6328464/mu.log)

carlosperate, 2021-05-11:
Hi @ldale15, I just wanted to check if you were able to try again with the latest beta release?

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-08-10 by carlosperate