mu-editor/mu #1484: Possible bug with PySimpleGUI import

SteveChoatie, 2021-04-17:
Hello all. 

I'm not sure if this is a bug or not. First let me tell you I am an absolute newbie to Python and I am trying to use the MU Editor. I am finding it a bit difficult to learn because it seems like in order to learn one thing I have to learn two more things and each one of those require me to learn other things and the thing keeps spiraling along.

I am using windows 10. and the code I'm running in Mu is pretty straightforward. It is: 

    import PySimpleGUI as sg
    sg.Window(title="Hello World", layout=[[]], margins=(100, 50)).read()

I receive the following error:
Traceback (most recent call last):
  File "c:\users\steve\mu_code\pysimple gui hello world.py", line 3, in <module>
    import PySimpleGUI as sg
ModuleNotFoundError: No module named 'PySimpleGUI'

I did a bunch of things trying to resolve the error such as making sure I installed PySimpleGui, updating pip, I updated Python tp 3.9.4, adding python to my path variable and testing the same code using Pycharm.  

I finally was able to get my code to work in Pycharm but not in Mu (version 1.0.3). I am at a loss at how to proceed. Below is a copy of my log file in Mu. 

Thanks!!

import PySimpleGUI as sg

sg.Window(title="Hello World", layout=[[]], margins=(100, 50)).read()
2021-04-17 13:37:54,140 - mu.logic:649(restore_session) INFO: Loaded files.
2021-04-17 13:37:54,140 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2021-04-17 13:37:54,140 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2021-04-17 13:37:54,159 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Steve\mu_code
2021-04-17 13:38:00,177 - mu.interface.panes:651(start_process) INFO: Running script: c:\users\steve\mu_code\pysimple gui hello world.py
2021-04-17 13:38:00,177 - mu.interface.panes:653(start_process) INFO: Running with interactive mode.
2021-04-17 13:38:00,177 - mu.interface.panes:656(start_process) INFO: Command args: []
2021-04-17 13:38:00,177 - mu.interface.panes:683(start_process) INFO: Python paths set via C:\Users\Steve\AppData\Roaming\Python\Python36\site-packages\mu.pth
2021-04-17 13:38:00,178 - mu.interface.panes:714(start_process) INFO: Working directory: C:\Users\Steve\mu_code
2021-04-17 13:38:00,178 - mu.interface.panes:719(start_process) INFO: Python path: ['C:\\Users\\Steve\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Steve\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Steve\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Steve\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Steve\\AppData\\Roaming\\Python\\Python36\\site-packages', 'C:\\Users\\Steve\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2021-04-17 13:38:13,002 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\Steve\AppData\Local\python\mu\Logs\mu.log

ajs256, 2021-04-18:
Mu uses its own virtual environment. Add the package name to the "Third-Party Packages" tab of the admin window to add it.
[image]

SteveChoatie, 2021-04-18:
Thank you very much!

SteveChoatie, 2021-04-18:
Hello again,
I attempted to follow your instructions about installing third party packages in Mu. However when I open up my version of Mu the tab you mentioned has been replaced (See below). Any idea where it has been moved to or how to do it otherwise? I checked the Mu documentation and the admin screen matches what I found. I also googled adding third party packages to Mu but wasn't successful in finding how to do it. 

Thanks,
Steve

ajs256, 2021-04-19:
Sorry, GitHub doesn't carry through attachments to email replies. Could you please attach it on the GitHub website [here](https://github.com/mu-editor/mu/issues/1484)? Thanks!

SteveChoatie, 2021-04-19:
I attempted to follow your instructions about installing third party packages in Mu. However when I open up my version of Mu the tab you mentioned has been replaced (See below). Any idea where it has been moved to or how to do it otherwise? I checked the Mu documentation and the admin screen matches what I found. I also googled adding third party packages to Mu but wasn't successful in finding how to do it.
[image]

Thanks,

Steve

SteveChoatie, 2021-04-19:
Thanks, I posted it on Github.

ajs256, 2021-04-19:
Make sure to choose "Python 3"
mode. Click the button in the top left with Mu's logo to switch.

SteveChoatie, 2021-04-19:
I did as you suggested and clicked on Mu's logo switch to choose Python 3. It was already in that mode so I checked the admin window again to see if I had the third packages tab. It still showed BBC mircrobit settings. So I changed to one of the other modes and then re-selected Python 3, closed Mu, restarted Mu and went to the admin window with the same results. Any thoughts?

Thanks

ajs256, 2021-04-19:
Hmm. Could a maintainer please step in here, because I have no clue what's going on here!

SteveChoatie, 2021-04-19:
Thanks for the additional help. I downloaded Mu-editor- Win64-1.1.04b.msi and installed it although a bit reluctantly. Windows defender indicated this could harm my computer but Norton said it was safe so I took a chance. After the installation I started Mu and received the attached crash report log.

Thanks,

Steve
[mu.log](https://github.com/mu-editor/mu/files/6338880/mu.log)
`

carlosperate, 2021-04-20:
Thanks for testing again @SteveChoatie!
To keep things separated I'll open a different issue to discuss the problem you've found with beta.4, as it looks different than the original problem discussed here. Once we have that sorted we can come back to this one.

https://github.com/mu-editor/mu/issues/1496

SteveChoatie, 2021-04-20:
I have some new information for all of you. Thank you for your help and pursuing this matter. I read the comments that you all have posted and most of them I did not understand since I'm new to Mu, Python, and a lot of this. I did come away with one thing from them, it seemed like the issue was related to old versions being left on the system. So I decided to try this. I copied my py exercise programs I had written and put them somewhere else. Then I uninstalled all Mu versions on my system beginning with the most current to the oldest version and then I reinstalled the latest release. When I opened Mu and went to the Administration tab I had a tab that said Third-Party Packages. I added the package I wanted, PySimpleGUI and tested my code. It worked!! So, my problem has been resolved on my end and hopefully this has helped you all with the beta version. Thank you again, I would have never figured it out on my own.

Steve Choatie

carlosperate, 2021-04-20:
I'm very glad you got it working!
I think the issue discussed here with PySinpleGUI has also been resolved by installing the dependency via the Mu settings, and the other issue from #1496 will be looked into as part of https://github.com/mu-editor/mu/issues/1421.

Thanks for all the testing!

closed 2021-04-20 by carlosperate