mu-editor/mu #1486: Crash Report

gadgetmuc, 2021-04-18:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Linux 5.4.0-65-generic #73-Ubuntu SMP Mon Jan 18 17:25:17 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: /home/dergadget/.local/share/mu/mu_venv-38-20210418-174045 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 162, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 596, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 649, in install_baseline_packages
    wheels.download()
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 79, in download
    process = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'download', '--destination-directory', '/usr/local/lib/python3.8/dist-packages/mu/wheels', 'pgzero>=1.2.1']' returned non-zero exit status 2.

```
[mu.log](https://github.com/mu-editor/mu/files/6331626/mu.log)

carlosperate, 2021-04-19:
Thanks for the report @gadgetmuc!

What version of Ubuntu are you running?

Could you try running the following two commands in a terminal and copy/paste the output here? That will help us with the debugging:

```
/usr/bin/python3 -m pip --version
```

```
/usr/bin/python3 -m pip download --destination-directory /usr/local/lib/python3.8/dist-packages/mu/wheels pgzero>=1.2.1
```

carlosperate, 2021-08-10:
Hi @gadgetmuc,

Would you be able to check if beta.5 works for you and let us know the result?
`pip install mu-editor==pip install mu-editor==1.1.0b5`

Thanks!

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate