mu-editor/mu #1488: crash report installing 1.1 beta 4

dsssssssss9, 2021-04-18:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)
OS: Ubuntu 20.04.2 LTS x86_64 

currently running 1.1 beta 3 with no problems.

attempted to install 1.1 beta 4 - which crashed after the animated snake screen with the attached log file

uninstalled beta 4 & reinstalled beta 3 for the time being
[mu.log](https://github.com/mu-editor/mu/files/6332185/mu.log)

carlosperate, 2021-04-19:
Thanks for the report and the log @dsssssssss9!

I believe the issue here might be https://github.com/mu-editor/mu/issues/1421. I think there might be two ways to work around this issue.

Option1:
- First `pip uninstall mu-editor`
- Check that the `/home/jkd/.local/lib/python3.8/site-packages/mu/` directory has been fully removed
   - if it's still present you can manually remove it
- Then `pip install mu-editor` again and this time it should work

Option 2:
- If you are currently running beta.3, then install beta.4 again
- Go to the `/home/jkd/.local/lib/python3.8/site-packages/mu/wheels` directory and remove all files **except** `__init__.py` and `__main__.py`. 
- Then run Mu and this time it should work
    - The first run will take a bit longer as it will redownload all the files you just deleted, but hopefully it'll fix the issue

dsssssssss9, 2021-04-19:
HI Carlos

thanks to you i am now looking at a working version of 1.1 beta 4 .

I am very grateful for the effort you took to help me out - particularly as you ( and the rest of the fantastic MU team! ) must be so so busy at the moment?

do you think this will be the best way to install future updates?

Regards

carlosperate, 2021-04-19:
I'm glad you were able to resolve the issue!
It's a busy time and we are all volunteers, but it's great to be able to able to work with a great community. Is thanks to people like you testing the editor, finding these issues, and investing the time to let us know, that we can do all this, so thank you for your time as well!

We'll be able to fix this issue for the next release, so this workaround should be only needed for beta.4, and future pip upgrades should work again 👍 

I'll close this issue as resolved, thanks for testing and don't hesitate to open more issues if you find other problems :)

closed 2021-04-19 by carlosperate