mu-editor/mu #1489: Cannot call pygame interface normally（No matter what I type or whether I type it or not）

planet618, 2021-04-19:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)
2021-04-19 10:03:38,993 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-04-19 10:03:38,993 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-AS6KGLU', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 12, GenuineIntel')
2021-04-19 10:03:38,993 - root:267(run) INFO: Python path: ['C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\YGQ\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-04-19 10:03:39,008 - root:268(run) INFO: Language code: zh_CN
2021-04-19 10:03:39,008 - mu.settings:220(load) WARNING: No settings file found at C:\Users\YGQ\AppData\Local\python\mu\settings.json; skipping
2021-04-19 10:03:39,008 - mu.settings:220(load) WARNING: No settings file found at C:\Users\YGQ\AppData\Local\python\mu\session.json; skipping
2021-04-19 10:03:39,357 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-04-19 10:03:39,588 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-04-19 10:03:39,588 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\YGQ\AppData\Local\python\mu\mu_venv-38-20210419-100328
2021-04-19 10:03:39,588 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210419-100328
2021-04-19 10:03:43,288 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-04-19 10:03:43,288 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\YGQ\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\YGQ\AppData\Local\python\mu\mu_venv-38-20210419-100328
2021-04-19 10:03:43,298 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\YGQ\AppData\Local\python\mu\mu_venv-38-20210419-100328\scripts\python.exe exists
2021-04-19 10:03:46,018 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: Requirement already satisfied: pip in c:\users\ygq\appdata\local\python\mu\mu_venv-38-20210419-100328\lib\site-packages (21.0.1)
2021-04-19 10:03:46,018 - mu.virtual_environment:669(upgrade_pip) INFO: Upgraded pip
2021-04-19 10:03:46,018 - mu.virtual_environment:711(install_baseline_packages) INFO: Installing baseline packages.
2021-04-19 10:03:46,018 - mu.virtual_environment:712(install_baseline_packages) INFO: C:\Users\YGQ\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-04-19 10:03:46,034 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-04-19 10:04:16,961 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ecdsa-0.16.1-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\esptool-3.0-py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\flask-1.1.2-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipykernel-5.5.3-py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython-7.22.0-py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ipython_genutils-0.2.0-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\itsdangerous-1.1.0-py2.py3-none-any.whl

Processing c:\users\ygq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\jedi-0.18.0-py2.py3-none-any.whl

[... 1342 lines cut ...]

pywin32                300

pyzmq                  22.0.3

qtconsole              4.7.4

QtPy                   1.9.0

reedsolo               1.5.4

setuptools             56.0.0

six                    1.15.0

tornado                6.1

traitlets              5.0.5

wcwidth                0.2.5

Werkzeug               1.0.1

wheel                  0.36.2
2021-04-19 10:47:07,764 - mu.virtual_environment:797(installed_packages) INFO: []

carlosperate, 2021-04-19:
Hi @planet618, thanks for the report.
We might need a bit more info about the issue, could you provide a bit more info about what you tried and what happened (or didn't happen)?

carlosperate, 2021-08-10:
Hi @planet618,

Do you have steps to reproduce this issue? We won't be able to help you without a way to reproduce it.

Thanks!

carlosperate, 2022-05-17:
Closing due to lack of activity.

Thanks for creating the report @planet618, and please do feel free to reopen the issue if you have some steps to reproduce the problem.

closed 2022-05-17 by carlosperate