mu-editor/mu #1492: Crash Report

pdalet, 2021-04-19:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b4
Locale: fr_FR
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Python37\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Python37\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Python37\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Python37\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Python37\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_key_modules
    "Failed to import: %s" % module
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: adafruit_board_toolkit

```

pdalet, 2021-04-19:
adafruit_board_toolkit is a tar.gz file on pypi.
I put the adafruit_board_toolkit-1.0.1-py3-none-any.whl on mu\wheels and it runs.
ph.D

closed 2021-04-19 by pdalet

carlosperate, 2021-04-20:
Thanks for the report @pdalet!
Did you install Mu on Windows via pip?
If that is the case, were you updating mu from a previous beta?

pdalet, 2021-04-21:
Le Tue, 20 Apr 2021 05:34:22 -0700,
Carlos Pereira Atencio ***@***.***> a écrit :

> Thanks for the report @pdalet!
> Did you install Mu on Windows via pip?
> If that is the case, were you updating mu from a previous beta?
> 

I have installed via pip and with a previous beta (b3) installed.

philippe

carlosperate, 2021-04-21:
Thanks for letting us know, that confirms this issue was due to https://github.com/mu-editor/mu/issues/1421. We'll track progress over there, so we can keep this issue closed.

Thanks for all the testing!