mu-editor/mu #1493: Crash Report

Mizan-github, 2021-04-19:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_AU
Platform: Windows 10 10.0.21359 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Mizan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Mizan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Mizan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Mizan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Mizan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

tjguk, 2021-04-20:
Thanks for the report, @Mizan-github -- would you be able to upload your log files to this ticket, please? You should find them at:

`%LOCALAPPDATA%\python\mu\logs\`

Best is to drop them onto the web interface -- they don't seem to come through to Github by email.

Mizan-github, 2021-04-20:
Thank you for getting back to me. The issue is sorted and it is working perfectly now.

Regards,Mizan

carlosperate, 2021-05-11:
Thanks for the report and the update @Mizan-github, in that case I'll close this issue 👍