mu-editor/mu #1494: Crash Report

FBSeletronica, 2021-04-20:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: pt_BR
Platform: Linux 5.11.0-7612-generic #13~1617215757~20.04~97a8d1a-Ubuntu SMP Thu Apr 1 21:15:20 UTC 2 x86_64

```
Traceback (most recent call last):
  File "/home/fabio/.local/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/fabio/.local/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/fabio/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "/home/fabio/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "/home/fabio/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: adafruit_board_toolkit

```

tjguk, 2021-04-20:
Thanks @FBSeletronica -- can I guess that you were upgrading from the previous beta? If so, you've hit a known bug which we are scrambling to fix :)

A workaround for you is to force Mu to recreate its runtime virtual environment. You can do this by finding the existing venv which should be in:

~/.cache/mu/mu_venv-<timestamp> 

and renaming it out of the way (or deleting it). Then start Mu again and it should detect the missing venv and recreate including the adafruit toolkit.

tjguk, 2021-04-20:
(Related to #1421)

carlosperate, 2021-04-20:
@tjguk as Mu will continue installing only the wheels in `mu/wheels/` and I suspect`adafruit_board_toolkit` is likely not there, I think it will continue to fail after deleting 

@FBSeletronica could you run this command on a terminal and copy/paste the output here?

```
ls /home/fabio/.local/lib/python3.8/site-packages/mu/wheels
```

Then, could you first try erasing `~/.cache/mu/mu_venv-*`, as Tim suggested, to see if that works on its own?

If that doesn't fix the issue could you try these steps?
- pip uninstall mu-editor
- Check if the `/home/fabio/.local/lib/python3.8/site-packages/mu/wheels` exists, if it does, delete it
- pip install mu-editor again

tjguk, 2021-04-20:
Thanks @carlosperate -- you're right; the presence of the existing whl files will prevent the new one(s) being added.

FBSeletronica, 2021-04-20:
@carlosperate  follow my terminal output

```
fabio@pop-os:~$ ls /home/fabio/.local/lib/python3.8/site-packages/mu/wheels
backcall-0.2.0-py2.py3-none-any.whl
bitstring-3.1.7-py3-none-any.whl
cffi-1.14.5-cp38-cp38-manylinux1_x86_64.whl
click-7.1.2-py2.py3-none-any.whl
cryptography-3.4.6-cp36-abi3-manylinux2014_x86_64.whl
decorator-4.4.2-py2.py3-none-any.whl
ecdsa-0.16.1-py2.py3-none-any.whl
esptool-3.0-py3-none-any.whl
Flask-1.1.2-py2.py3-none-any.whl
__init__.py
ipykernel-5.5.0-py3-none-any.whl
ipython-7.21.0-py3-none-any.whl
ipython_genutils-0.2.0-py2.py3-none-any.whl
itsdangerous-1.1.0-py2.py3-none-any.whl
jedi-0.18.0-py2.py3-none-any.whl
Jinja2-2.11.3-py2.py3-none-any.whl
jupyter_client-6.1.12-py3-none-any.whl
jupyter_core-4.7.1-py3-none-any.whl
__main__.py
MarkupSafe-1.1.1-cp38-cp38-manylinux2010_x86_64.whl
nudatus-0.0.5-py3-none-any.whl
numpy-1.20.1-cp38-cp38-manylinux2010_x86_64.whl
parso-0.8.1-py2.py3-none-any.whl
pexpect-4.8.0-py2.py3-none-any.whl
pgzero-1.2-py3-none-any.whl
pickleshare-0.7.5-py2.py3-none-any.whl
prompt_toolkit-3.0.18-py3-none-any.whl
ptyprocess-0.7.0-py2.py3-none-any.whl
__pycache__
pycparser-2.20-py2.py3-none-any.whl
pygame-2.0.1-cp38-cp38-manylinux1_x86_64.whl
Pygments-2.8.1-py3-none-any.whl
pyserial-3.5-py2.py3-none-any.whl
python_dateutil-2.8.1-py2.py3-none-any.whl
pyzmq-22.0.3-cp38-cp38-manylinux2010_x86_64.whl
qtconsole-4.7.4-py2.py3-none-any.whl
QtPy-1.9.0-py2.py3-none-any.whl
reedsolo-1.5.4-py3-none-any.whl
setuptools-54.2.0-py3-none-any.whl
six-1.15.0-py2.py3-none-any.whl
tornado-6.1-cp38-cp38-manylinux2010_x86_64.whl
traitlets-5.0.5-py3-none-any.whl
wcwidth-0.2.5-py2.py3-none-any.whl
Werkzeug-1.0.1-py2.py3-none-any.whl
```

@carlosperate and @tjguk 

I had to uninstall and re-install mu. But pip install mu-editor doesn't work

```
fabio@pop-os:~$ pip3 install mu-editor
Collecting mu-editor
  Using cached mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
Collecting pyqt5==5.14.1
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
Requirement already satisfied: nudatus>=0.0.3 in ./.local/lib/python3.8/site-packages (from mu-editor) (0.0.5)
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Requirement already satisfied: appdirs>=1.4.3 in ./.local/lib/python3.8/site-packages (from mu-editor) (1.4.4)
Collecting PyQtChart==5.14.0
  Using cached PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
Collecting pigpio>=1.40.post1
  Using cached pigpio-1.78-py2.py3-none-any.whl (39 kB)
Collecting pyflakes==2.0.0
  Using cached pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
[... 59 lines cut ...]
                       agg: yes [installing]
                     cairo: yes [installing, pycairo version 1.16.2]
                 windowing: no  [Microsoft Windows only]
    
    OPTIONAL LATEX DEPENDENCIES
                    dvipng: no
               ghostscript: yes [version 9.50]
                     latex: no
                   pdftops: yes [version 0.86.1]
    
    OPTIONAL PACKAGE DATA
                      dlls: no  [skipping due to configuration]
    
    ============================================================================
                            * The following required packages can not be built:
                            * freetype, png
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

I had to install the version mu-editor==1.1.0b4 to work.

[image]

Thanks you guys.

carlosperate, 2021-04-20:
Thanks @FBSeletronica, I forgot the pip install has to specify the version or it will try to install v1.0.3.

And I'm glad you got it working! I'll close this issue and we'll track the work require to fix the problem in https://github.com/mu-editor/mu/issues/1421.

Thanks again for all the testing!

closed 2021-04-20 by carlosperate

FBSeletronica, 2021-04-20:
Thank you @carlosperate