mu-editor/mu #1495: Submit final installers to antiviruses allowlists

carlosperate, 2021-04-20:
There are reports of some antiviruses warning against the Mu installer: https://github.com/mu-editor/mu/issues/1484#issuecomment-822764888

We'll probably need to submit the final installers for consideration in their allowlists or check what we might need to do to not be considered "dangerous".

I believe @ntoll has done something related to this for Norton, and we looked into issues with McAfee in the past: https://github.com/mu-editor/mu/issues/280

tjguk, 2021-04-20:
@carlosperate  -- I'm fairly sure that the "Windows defender" issue is basically the fact that the installer isn't signed (because @ntoll can't yet renew the Windows cert). It's the MS equivalent of the Apple "this app can seriously damage your health" warning which we had to cope with on OSX.

carlosperate, 2021-04-20:
I though the Windows installer was fully signed now? Is that not the case?

carlosperate, 2021-08-04:
Moving the conversation from https://github.com/mu-editor/mu/issues/1600#issuecomment-877709029 here:

> Despite many efforts I can't get this to download. Microsoft Defender blocks the msi file as unsafe (so I can't even try to install it). I have reported it as a safe file so hopefully this will be resolved quickly.

@AnnEvans is there no way to bypass that warning?
Would you be able to take any screenshots trying to download the file https://github.com/mu-editor/mu/releases/download/1.1.0-beta.5/Mu-Editor-Win64-1.1.0b5.msi ?

carlosperate, 2021-08-13:
Getting this warning on Edge browser on Windows, but not on macOS:

[image]

The "see more" link/button doesn't do anything, I think that's only to see previously downloaded files in the list.

When mouse over the entry this is the menu from the ellipsis button:
[image]

We need users to click on "keep" to be able to download the file.
Ideally we could ask people to "Report this file as safe" first, that might be able to build some kind of trust over time.

@ntoll there is an option to submit a form as the owner, so that this warning/blockage can get reviewed:
[image]

carlosperate, 2021-08-13:
After all that we also get this "Windows Defender Smart screen":
[image]

carlosperate, 2021-09-03:
Just FYI, Firefox will soon block non-signed executable downloads as well: https://therecord.media/firefox-follows-chrome-and-prepares-to-block-insecure-downloads/

carlosperate, 2022-02-25:
The Edge issue was due to the Windows installer not being signed, now that they are signed again the warnings do not trigger any more.

The antivirus issue looks like might not be a problem anymore, if a user reports an antivirus marking Mu as a thread we can look into it at that point.

This is good to be closed 👍

closed 2022-02-25 by carlosperate