mu-editor/mu #1496: Crash report - ResolutionImpossible

carlosperate, 2021-04-20:
From https://github.com/mu-editor/mu/issues/1484#issuecomment-822764888:

```
2021-04-19 15:23:18,850 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
```

Log: [mu.log](https://github.com/mu-editor/mu/files/6338880/mu.log)

beta.4 on Windows.

carlosperate, 2021-04-20:
From the log file:

```
ERROR: Cannot install decorator 4.4.2 (from C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\decorator-4.4.2-py2.py3-none-any.whl) and decorator 5.0.7 (from C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl) because these package versions have conflicting dependencies.

The conflict is caused by:

    The user requested decorator 4.4.2 (from C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\decorator-4.4.2-py2.py3-none-any.whl)

    The user requested decorator 5.0.7 (from C:\Users\Steve\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl)

To fix this you could try to:

1. loosen the range of package versions you've specified

2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies
```

Looks like Mu has downloaded two `decorator` wheels at different versions, I would guess this might be an issue that would affect all Windows users? @tjguk

tjguk, 2021-04-20:
Interesting; I assume it's an issue when overinstalling from a previous (Mu) version. Otherwise I would have expected to hit the problem during testing.

carlosperate, 2021-04-20:
I think you are right, looking at the macOS installers v4.4.2 is from beta.3 and v5.0.7 from beta.4.

Should the installer be configured to remove old files, or we want to do these checks as part of the mu/wheels code?

tjguk, 2021-04-20:
I think we'd better put whatever logic into the mu/wheels code. Need to come up with some version-check / upgrade logic

carlosperate, 2021-04-21:
Okay, as we are tracking this issue in https://github.com/mu-editor/mu/issues/1421 I'll close this ticket.

closed 2021-04-21 by carlosperate

SteveChoatie, 2021-04-21:
I was happy to help with the testing and glad my question ended up helping you all out. Thanks so much for your efforts!