mu-editor/mu #1498: Where us the .exe file to open the mu thing?

Naruhota, 2021-04-20:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2021-08-17:
Hi @Naruhota,

There isn't a specific .exe that can be executed as is, as Mu packs it's own Python interpreter and then it uses it to launch a Python module (Mu itself).

The Windows installer should have added a shortcut to the start menu, were you able to find it?

carlosperate, 2022-05-17:
As there wasn't a response I assume your question was answer, so I'll close this issue. But please feel free to reopen if you still have any questions.

closed 2022-05-17 by carlosperate