mu-editor/mu #1500: Crash Report

soretoez, 2021-04-22:
[mu.log](https://github.com/mu-editor/mu/files/6360980/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6360982/mu.log)

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

carlosperate, 2021-05-06:
Error from the log:

```
2021-04-22 21:09:00,074 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210256\scripts\python.exe exists
2021-04-22 21:09:00,315 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 101; output: Unable to create process using 'C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\python.exe -m pip install --upgrade pip'
2021-04-22 21:09:00,315 - mu.virtual_environment:464(quarantine_failed_venv) ERROR: Unable to quarantine C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210256 as C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210256.FAILED
Traceback (most recent call last):
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 462, in quarantine_failed_venv
    os.rename(self.path, error_dirpath)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\jason hart\\AppData\\Local\\python\\mu\\mu_venv-38-20210422-210256' -> 'C:\\Users\\jason hart\\AppData\\Local\\python\\mu\\mu_venv-38-20210422-210256.FAILED'
2021-04-22 21:09:00,330 - mu.virtual_environment:410(relocate) DEBUG: Virtual environment set up mu_venv-38-20210422-210900 at C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210900
2021-04-22 21:09:00,330 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x00000150AAB5F280>; Splash Handler; <SplashLogHandler (INFO)>
2021-04-22 21:09:00,330 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #2.
2021-04-22 21:09:00,330 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210900
2021-04-22 21:09:00,330 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210422-210900
2021-04-22 21:09:01,643 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-04-22 21:09:01,643 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210900
2021-04-22 21:09:01,643 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210900\scripts\python.exe exists
2021-04-22 21:09:01,920 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 101; output: Unable to create process using 'C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\python.exe -m pip install --upgrade pip'
2021-04-22 21:09:01,920 - mu.virtual_environment:468(quarantine_failed_venv) INFO: Quarantined C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210900 as C:\Users\jason hart\AppData\Local\python\mu\mu_venv-38-20210422-210900.FAILED
2021-04-22 21:09:01,920 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x00000150AAB5F280>; Splash Handler; <SplashLogHandler (INFO)>
2021-04-22 21:09:08,936 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\jason hart\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
```

Looks like we have the "Unable to upgrade pip error" and also a permission error trying to rename a venv folder that has failed the checks:
https://github.com/mu-editor/mu/blob/2b20da24e5219002fe19781509fa8665a9842464/mu/virtual_environment.py#L456-L463

@tjguk I guess this might be permission issues?
I could also explain issues https://github.com/mu-editor/mu/issues/1506#issuecomment-827072330 and https://github.com/mu-editor/mu/issues/1506#issuecomment-827072330, where the user has no permissions so `os.path.exists` returns False and then it fails to create the file/directory.

carlosperate, 2021-08-06:
Hi @soretoez!

The beta.5 has some fixes in this area, would you be able to install this release and try again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-12-17:
Hi @soretoez, are you still experiencing this problem?

Are you able to test the latest beta release to see if that resolved the issue?
https://github.com/mu-editor/mu/releases/

carlosperate, 2022-05-31:
Closing due to lack of activity.