mu-editor/mu #1506: IDK what happened yall better not be trolling and steal all my data ill sue u ppl and call the cops

Oui-Oui-Oui-Oui, 2021-04-25:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: zh_MO
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, 
[mu.log](https://github.com/mu-editor/mu/files/6372158/mu.log)
[None,]
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 344, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 829, in setup
    shutil.copytree(path("templates", "web/"), template_path)
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 554, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 455, in _copytree
    os.makedirs(dst, exist_ok=dirs_exist_ok)
  File "C:\Users\kioki\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] ç¶æªæ¡å·²å­å¨æï¼ç¡æ³å»ºç«è©²æªæ¡ã: 'C:\\Users\\kioki\\mu_code\\templates'

```

carlosperate, 2021-04-26:
Rest assured @Oui-Oui-Oui-Oui, this bug report you've open contains some info about the Mu editor crashing that can help us ensure we can fix it in future version.

As this looks like it might be a bug in Mu I'll reopen so that we can look into it:
https://github.com/mu-editor/mu/blob/38f905a2fcb57b5438a39862b8823c31e74e0290/mu/logic.py#L827-L829

It's a bit weird that line 827 checks the directory does **not** exist but it somehow materialised by the time it reaches line 829?

Looking at the docs `os.exists()` could erroneously return `False` if permission is not granted to execute `os.stat()` on the requested file: https://docs.python.org/3/library/os.path.html#os.path.exists

But then `os.mkdir()` in `shutil.copytree()` throws the `FileExistsError`. @tjguk would it be possible in Windows for the user to not have permissions to the mu_code folder, where `os.exists()` doesn't "see" the directory but `os.mkdir()` does? Or am I barking at the wrong tree here?

carlosperate, 2021-12-17:
Right before the error we can see duplicated entries in the log:

```
2021-04-25 22:21:34,540 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\kioki\mu_code from settings file
2021-04-25 22:21:34,540 - mu.logic:812(setup) DEBUG: Creating directory: C:\Users\kioki\mu_code\fonts
2021-04-25 22:21:34,541 - mu.logic:815(setup) DEBUG: Creating directory: C:\Users\kioki\mu_code\sounds
2021-04-25 22:21:34,588 - mu.logic:822(setup) DEBUG: Creating directory: C:\Users\kioki\mu_code\music
2021-04-25 22:21:34,589 - mu.logic:828(setup) DEBUG: Creating directory: C:\Users\kioki\mu_code\templates
2021-04-25 22:21:34,589 - mu.logic:828(setup) DEBUG: Creating directory: C:\Users\kioki\mu_code\templates
2021-04-25 22:21:34,589 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
```

So this crash was caused by issue #1755.

Assuming that's the case Mu probably run correctly the next time you tried to launch it, is that correct @Oui-Oui-Oui-Oui ? Or are you still experiencing this problem?

carlosperate, 2022-01-19:
As this problem is captured in #1755, I'll close this issue to keep things tidy.
Thanks for the report!

closed 2022-01-19 by carlosperate