mu-editor/mu #1507: now f

Sn8ow, 2021-04-26:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2021-04-26:
Hi @Sn8ow, it looks like maybe the content of your bug report didn't reach us. Did you have any issues with the Mu editor you would like to report?

Sn8ow, 2021-04-27:
No I'm sorry

carlosperate, 2021-04-27:
No worries, I'll close this issue then 👍

closed 2021-04-27 by carlosperate