mu-editor/mu #1508: Mu log is printed to the terminal

carlosperate, 2021-04-26:
Running Mu from source code, or from the PyPI package, and all the log is being outputted to the terminal as well, which can be quite verbose. I assume this change was not by design?

Tested in macOS Big Sur and Ubuntu 21.04, running Mu from the terminal via `$ python run.py` or `$ mu-editor`.

tjguk, 2021-04-26:
Sorry; that would have been me. I normally have it commented out but obviously I enabled it for something and forgot to disable again. I'll have to come up with some env-var or something which lets me turn it on when I need it.

carlosperate, 2021-04-26:
No worries, an env-var for that might be a good idea in general, definitely something I can see myself use too.

carlosperate, 2021-06-14:
This was originally implemented in https://github.com/mu-editor/mu/pull/1530/ but got lost in the PR division, raised PR https://github.com/mu-editor/mu/pull/1640 to add it again.

closed 2021-06-19 by ntoll