mu-editor/mu #1511: Commenting (CTRL-K) deletes first character in the next line

CsabaTizedes, 2021-04-27:
If you are reporting a bug, we would like to know:

* What you were trying to do,
From this code I tried to **comment the first `pixel.show()`**
`pixel[1] = (255, 255, 255)`
`pixel.show()`
`pixel[2] = (255, 255, 255)`
`pixel.show()`

* What steps you took to make this happen,
I placed the cursor at the beginning of the line **without highlighting** the command and I pressed CTRL-K

* What you expected to happen,
The line will be commented, like this:
`# pixel.show()`

* What actually happened,
The line is commented and highlighted, but the cursor placed into the next line and the **next line first character is deleted**:
`pixel.show()`
`ixel[2] = (255, 255, 255)`

**IF the line is highlighted before CTRL-K, there is no any problem.**

* Why this difference is problematic (it may not be a bug!),
Because one character is deleted ...

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
MU version: 1.1.0.beta.2
OS: Windows 10 Pro, version 2004

Please remember to attach a **copy of the full log files for Mu**. 
The log is attached.
[log_Commenting (CTRL-K) deletes first character in the next line_Tizedes.txt](https://github.com/mu-editor/mu/files/6382285/log_Commenting.CTRL-K.deletes.first.character.in.the.next.line_Tizedes.txt)

carlosperate, 2021-04-27:
Hi @CsabaTizedes, I think I'm unable to replicate this issue in macOS. Could you have a look at this video to see if I have understood the steps correctly?

https://user-images.githubusercontent.com/4189262/116211006-4d7ed980-a73b-11eb-9669-b495159f973e.mp4

It might be possible this could be an issue due to some unicode characters present somewhere else in the script. Could you copy the full script here?

Also, are you able to replicate in a new tab with only those 4 lines?
```
pixel[1] = (255, 255, 255)
pixel.show()
pixel[2] = (255, 255, 255)
pixel.show()
```

CsabaTizedes, 2021-04-27:
Hi @carlosperate, 

In new tab I tried those four lines and the result is same. Please check the video here: [LINK](https://youtu.be/pVP8FWm9-Ek)
(I'm new here, I hope the link from YouTube is accepted and stay here).

carlosperate, 2021-04-27:
Yeah, videos in any form are really useful to understand the issue, thanks @CsabaTizedes!

Interesting, I guess we'll have to try this on Windows to replicate.
I suspect this might be a line ending issue?

devdanzin, 2021-12-04:
Confirmed and it's a line ending issue indeed, that's why it needs to be a new tab (otherwise line endings seem to be set to `"\n"`). Doesn't seem to happen when text is previously selected. When text isn't selected, we select it to replace and an off-by-one error selects the first character on next line.
The code we hit is:

```python
            line_number, column = self.getCursorPosition()
            logger.info("Toggling line {}".format(line_number))
            line_content = self.text(line_number)
            new_line = self.toggle_line(line_content)
            self.setSelection(line_number, 0, line_number, len(line_content))
            self.replaceSelectedText(new_line)
            self.setSelection(line_number, 0, line_number, len(new_line) - 1)
```
... and removing `"\r"` like `line_content = self.text(line_number).replace("\r", "")` seems to fix it. Still have to figure out a test and check this doesn't mess the file with inconsistent line endings.

commit a947a658 (2021-12-20): Merge pull request #1911 from devdanzin/fix_toggle_deletes

Fix deleting the first character of next line when toggling comment (Ctrl+K). Fixes #1511