mu-editor/mu #1512: Crash Report

bajok062, 2021-04-27:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: nb_NO
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\BÃ¥rd JÃ¸rgen Kollbotn\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\BÃ¥rd JÃ¸rgen Kollbotn\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\BÃ¥rd JÃ¸rgen Kollbotn\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\BÃ¥rd JÃ¸rgen Kollbotn\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\BÃ¥rd JÃ¸rgen Kollbotn\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in upgrade_pip
    ok, output = self.run_subprocess(
  File "C:\Users\BÃ¥rd JÃ¸rgen Kollbotn\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 382, in run_subprocess
    compact(process.stdout.decode("utf-8")),
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe5 in position 48: invalid continuation byte

```

carlosperate, 2021-04-27:
Thanks for the report @bajok062!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

bajok062, 2021-04-28:
[mu.log](https://github.com/mu-editor/mu/files/6389854/mu.log)

bajok062, 2021-04-29:
The installation prosess is not crashing  with this mu-version: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.3

1.1.0-beta.3

carlosperate, 2021-04-29:
Thanks for that clarification, that's definitely useful to pin the change to beta.4.

@tjguk looks like some of the latest venv changes might need to detect the subprocess stdout encoding? Based on the error it looks like it might not be UTF-8?

tjguk, 2021-04-29:
Thanks @carlosperate  (and @bajok062  for reporting). Yes; I was a bit optimistic by just throwing utf-8 at that :)

I'll try to detect the encoding and put in a fallback just in case

carlosperate, 2021-04-29:
I don't think we are currently using `chardet` in the Mu source code, but it is brought in as a subdependency from somewhere, so it wouldn't inflate the package if we include it in the `instal_requires`.
If we decide to use it one thing to keep in mind is that v4 is 3.6+ only, but we should be able to stick with the latest v3 even if it's a bit old.

carlosperate, 2021-05-05:
Merging this crash report here to have a single place to track this issue: https://github.com/mu-editor/mu/issues/1534

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: pt_BR
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\FamÃ­lia Santos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\FamÃ­lia Santos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\FamÃ­lia Santos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\FamÃ­lia Santos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\FamÃ­lia Santos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in upgrade_pip
>     ok, output = self.run_subprocess(
>   File "C:\Users\FamÃ­lia Santos\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 382, in run_subprocess
>     compact(process.stdout.decode("utf-8")),
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xed in position 50: invalid continuation byte
> ```

@gabriel4569 if you could attach your log to here it would be quite useful too, as that will help us gather more information on the issue. The log file should be located in: `%LOCALAPPDATA%\python\mu\Logs`

tjguk, 2021-05-05:
Thanks, @carlosperate -- I have already implemented what should be a fix for this as part of #1530. I'll add tests over there.

carlosperate, 2021-05-05:
Merging this crash report here to have a single place to track this issue: https://github.com/mu-editor/mu/issues/1537

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_GB
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\IvÃ¡n\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\IvÃ¡n\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\IvÃ¡n\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\IvÃ¡n\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\IvÃ¡n\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in upgrade_pip
>     ok, output = self.run_subprocess(
>   File "C:\Users\IvÃ¡n\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 382, in run_subprocess
>     compact(process.stdout.decode("utf-8")),
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe1 in position 49: invalid continuation byte
> ```

@TreintaTreinta I've merged your issue here to track this in a single place, thanks for the report!

carlosperate, 2021-06-07:
Merging this crash report here to have a single place to track this issue: https://github.com/mu-editor/mu/issues/1623

> Version: 1.1.0.beta.4
> Locale: de_CH
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\RenÃ© HÃ¼rzeler\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\RenÃ© HÃ¼rzeler\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\RenÃ© HÃ¼rzeler\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\RenÃ© HÃ¼rzeler\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\RenÃ© HÃ¼rzeler\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in upgrade_pip
>     ok, output = self.run_subprocess(
>   File "C:\Users\RenÃ© HÃ¼rzeler\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 382, in run_subprocess
>     compact(process.stdout.decode("utf-8")),
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe9 in position 50: invalid continuation byte
> ```

@rhuerzeler  I've merged your issue here to track this in a single place, thanks for the report!

tjguk, 2021-06-07:
I'm fairly sure we've got this covered now by changes in #1622 -- but I do need to add some tests.

carlosperate, 2021-08-04:
@bajok062 @gabriel4569 @TreintaTreinta @rhuerzeler

Hi everyone! 👋

The issues you were having with Mu should have been resolved with the beta.5 release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Would you be able to try the installer from this release and check if it is working now?

Thanks everyone for your help testing the Mu Editor!

carlosperate, 2021-09-22:
We haven't received any reports about this issue since beta.5 was released, so I'll consider this resolved.
Thanks everybody!

closed 2021-09-22 by carlosperate