mu-editor/mu #1513: Crash Report

prefontdon, 2021-04-27:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_CA
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\prefo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\prefo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\prefo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\prefo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\prefo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

prefontdon, 2021-04-27:
.

carlosperate, 2021-04-27:
Hi @prefontdon, thanks for the report!
As this looks like an issue starting up Mu I'll reopen so that we can look into it.

Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

prefontdon, 2021-04-27:
Removed app from PC.

parsko41, 2021-04-27:
[mu.log](https://github.com/mu-editor/mu/files/6387325/mu.log)
I have had it fail too.  I've got AVG installed, and added some folders to my exception list, and I think that allowed it to move forward until it got to pgzero.

-Parsko

carlosperate, 2021-04-28:
Hi @parsko41,

I think your issue might be different, and probably https://github.com/mu-editor/mu/issues/1421.

Could you follow these steps to see if that resolves it?
- uninstall mu-editor
- Check that the `C:\Users\renamed\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\` folder no longer exists
    - if it's still present you can manually remove it
- Install mu-editor again and this time it should work

prefontdon, 2021-04-28:
Carlos

The folder/directory did not exist.
I reinstalled it and it works fine.

Thank You,

Don

carlosperate, 2021-04-28:
I'm very glad to hear that resolved your issue @prefontdon! Thanks for testing again and the update.

@parsko41 if you could try those steps as well hopefully it'll resolve your issue too.
We'll look into fixing this problem for the next release.

Thanks again for the reports and testing!

parsko41, 2021-04-28:
@carlosperate 

Yes, your advice fixed my issue.  I still had older versions installed as well.  Uninstalling and starting fresh resolved it.  Always the best choice with software. :)

Thanks.

prefontdon, 2021-04-28:
I neglected to mention that I had an older version installed, too. With
both gone and then the newest installed, it worked fine.

Thanks.

carlosperate, 2021-04-28:
Thanks both for the update! I'm glad you were able to resolve it with the "uninstall first" workaround, that definitely confirms this was due to #1421.
We'll track the work to fix this in #1421, so I'll close this issue. Thanks for all the testing, it's really appreciated!

closed 2021-04-28 by carlosperate