mu-editor/mu #1514: Crash Report

gustavo123-pixel, 2021-04-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_NZ
Platform: Windows 10 10.0.17134 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Gustavo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Gustavo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Gustavo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Gustavo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 623, in create
    self.create_venv()
  File "C:\Users\Gustavo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\Gustavo\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\Gustavo\AppData\Local\python\mu\mu_venv-38-20210428-144242

```

carlosperate, 2021-06-04:
HI @gustavo123-pixel, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\Gustavo\AppData\Local\python\mu\logs`.

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-08-10 by carlosperate