mu-editor/mu #1515: Crash Report

BarryButler, 2021-04-28:
**REMEMBER TO ATTACH YOUR LOG FILE**
I have searched the C: and D: drives to locate the mu.log file and can find no trace of it.  The crash report says it should be located at C:\Users\bbutl58\AppData\Local\python\mu\Logs\mu.log, but there is no AppData\Local folder.  The installation failed as below.  I am running python 3.7.0

Version: 1.1.0.beta.4
Locale: en_AU
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\bbutl58\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\bbutl58\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\bbutl58\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\bbutl58\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\bbutl58\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-04-28:
Hi @BarryButler, thanks for the report!

By default the AppData folder will be hidden, but you can still access it by entering the location in a explorer window address bar. Hopefully this link will explain it a bit better than me 😅
https://help.ableton.com/hc/en-us/articles/209070509-How-to-access-hidden-folders#windows10

Let me know if you have any questions and thanks again for sending the report, these really help us find and fix bugs in Mu.

carlosperate, 2021-06-04:
Hi @BarryButler, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\bbutl58\AppData\Local\python\mu\logs`.

carlosperate, 2021-08-10:
Hi @BarryButler 

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@BarryButler have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5