mu-editor/mu #1516: Crash Report

LionSoft91, 2021-04-28:
[mu.log](https://github.com/mu-editor/mu/files/6393191/mu.log)
Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l
[mu.log](https://github.com/mu-editor/mu/files/6393181/mu.log)

```
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 109, in start_kernel
    self.repl_kernel_manager.start_kernel()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 42, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
AttributeError: 'MuKernelManager' object has no attribute 'pre_start_kernel'

```

carlosperate, 2021-04-28:
Thanks for the report @LionSoft91, this might be related to https://github.com/mu-editor/mu/issues/1444.

Could you run this command (with the same pip command you've used to install Mu) in the terminal and copy/paste the result here?

```
pip3 list
```

LionSoft91, 2021-04-28:
This is what I got by using the command :

```
Package                  Version    
------------------------ -----------
adafruit-board-toolkit   1.0.1      
appdirs                  1.4.3      
asn1crypto               0.24.0     
astroid                  2.1.0      
asttokens                1.1.13     
automationhat            0.2.0      
beautifulsoup4           4.7.1      
black                    21.4b1     
blinker                  1.4        
blinkt                   0.1.2      
buttonshim               0.0.2      
cachetools               4.2.2      
Cap1xxx                  0.1.3      
certifi                  2018.8.24  
chardet                  3.0.4      
click                    7.1.2      
colorama                 0.3.7      
colorzero                1.1        
cookies                  2.2.1      
cryptography             2.6.1      
cupshelpers              1.0        
cycler                   0.10.0     
decorator                4.3.0      
distlib                  0.3.1      
docutils                 0.14       
drumhat                  0.1.0      
entrypoints              0.3        
envirophat               1.0.0      
et-xmlfile               1.1.0      
ExplorerHAT              0.4.2      
EZGmail                  2021.2.4   
EZSheets                 2020.10.10 
filelock                 3.0.12     
flake8                   3.9.1      
Flask                    1.0.2      
fourletterphat           0.1.0      
google-api-core          1.26.3     
google-api-python-client 2.2.0      
google-auth              1.30.0     
google-auth-httplib2     0.1.0      
google-auth-oauthlib     0.4.4      
googleapis-common-protos 1.53.0     
gpiozero                 1.5.1      
guizero                  0.6.0      
html5lib                 1.0.1      
httplib2                 0.19.1     
idna                     2.6        
IMAPClient               2.1.0      
importlib-metadata       4.0.1      
ipykernel                4.9.0      
ipython                  5.8.0      
ipython-genutils         0.2.0      
isort                    4.3.4      
itsdangerous             0.24       
jdcal                    1.4.1      
jedi                     0.13.2     
Jinja2                   2.10       
jupyter-client           5.2.3      
jupyter-core             4.4.0      
keyring                  17.1.1     
keyrings.alt             3.1.1      
kiwisolver               1.0.1      
lazy-object-proxy        1.3.1      
logilab-common           1.4.2      
lxml                     4.3.2      
[... 93 lines cut ...]
sn3218                   1.2.7      
soupsieve                1.8        
spidev                   3.4        
ssh-import-id            5.7        
thonny                   3.3.6      
toml                     0.10.2     
tornado                  5.1.1      
touchphat                0.0.1      
traitlets                4.3.2      
twilio                   6.57.0     
twython                  3.7.0      
typed-ast                1.4.3      
typing-extensions        3.7.4.3    
uflash                   1.2.4      
unicornhathd             0.0.4      
uritemplate              3.0.1      
urllib3                  1.24.1     
virtualenv               20.4.4     
wcwidth                  0.1.7      
webencodings             0.5.1      
Werkzeug                 0.14.1     
wheel                    0.32.3     
wrapt                    1.10.11    
zipp                     3.4.1  
```

carlosperate, 2021-04-28:
Thanks @LionSoft91, that was really useful!

Ultimately it looks like in PR https://github.com/mu-editor/mu/pull/1240 we started using a `jupyter_client` method (`KernelManager.pre_start_kernel()`) introduced in v6.1.0, which we are not enforcing in our `install_requires`.

I'll raise a PR to fix this.

LionSoft91, 2021-04-28:
Thanks @carlosperate for your feedback.

Do you know an alternative solution to use REPL because I started to learn python following a book called "**AUTOMATE THE BORING STUFF WITH PYTHON, 2ND EDITION Practical Programming for Total Beginners**" by "**AL SWEIGART**" and it uses REPL on MU.

carlosperate, 2021-04-28:
The easiest way to fix this would be to update the `jupyter_client` package to a newer version. However, since the reason you have an old version is because some other application in your computer is using it, there is a risk that updating it could cause issues in that other application.

You can install Mu in a virtual environment, this isolates the Mu installation, so that its dependencies are not combined with the system packages. 
The main inconvenience from using a virtual environment is that it has to be "installed" somewhere (a folder where all the dependencies are stored), and then every time you want to use Mu you'll have to "activate" it.

These are the steps:
1) In a terminal, navigate to a directory where you want to install the virtual environment
2) Run this to create the virtual environment in a folder named `mu-venv`: `virtualenv --python=python3 mu-venv`
3) Run this to activate it: ` source mu-venv/bin/activate`
4) Run this to install Mu in the virtual environment: `pip install mu-editor==1.1.0b4`
5) Run Mu from the virtual environment: `mu-editor`

Every time you want to run Mu you'd have to do steps 1, 3, and 5 (2 and 4 are only required to install it).

Alternatively I like to use the pipx tool: https://pipxproject.github.io/pipx/
If you install it you can then use pipx to automatically install Python applications in their own isolated virtual environment, and they can be run without manually activating the environment:
- Install pipx
    - Update pip: `python3 -m pip install --upgrade pip` 
    - Run: `python3 -m pip install --user pipx`
    - Run: `python3 -m pipx ensurepath`
- After installation you might need to close all your terminals and open a new one (or maybe even reboot, the last command should print out information)
- Now that is pipx is installed and configured you can use it to install Mu: `pipx install mu-editor==1.1.0b4`
- And you can run mu: `mu-editor`

carlosperate, 2021-06-23:
Hi @LionSoft91, I was wondering if you managed to resolve this issue?

LionSoft91, 2021-06-23:
Hello,
Yes, it resolved with the update.
Thank you.

LionSoft91, 2021-06-23:
> Hi @LionSoft91, I was wondering if you managed to resolve this issue?

Hello,
Yes, it resolved with the update.
Thank you.

carlosperate, 2021-06-23:
That's great, thanks for the update!
In that case I'll close this issue, thanks for all the testing and please don't hesitate to open another issue if you find any other problems 👍

closed 2021-06-23 by carlosperate