mu-editor/mu #1521: docs: explenation of Linux dependencys missing

newsgrep, 2021-04-30:
Please make the documentation on how to setup mu on Linux more verbose. 
https://codewith.mu/en/howto/1.1/install_with_python
* there is no explanation of the needed dependency's or how to install those. E.g. libsdl1.2-dev for Ubuntu Linux (Ubuntu is often used by people new to Linux)
* please be a bit more verbose regarding the usage of virtualenv

I mean you explain what a command prompt so I think it would be appropriate to explain the thins above more.

carlosperate, 2021-09-21:
Hi @newsgrep,

`sudo pip3 install mu-editor --pre` should have achieved the same as `sudo pip3 install mu-editor==1.1.0b5`. Did you specify the version in the pip install command?

You shoudn't need to install any other Debian/Ubuntu packages like libsdl1.2-dev. What version of Ubuntu were you using? It's possible it was trying to build some wheels from source (like pygame), which shouldn't be the case for any major Linux distro with the latest Mu editor beta.

For the venv documentation we have this issue to track that: https://github.com/mu-editor/mu/issues/1328

carlosperate, 2022-05-17:
We are tracking the doc updates required in https://github.com/mu-editor/mu/issues/1328, so we can close this issue. Thanks @newsgrep for raising it!

closed 2022-05-17 by carlosperate