mu-editor/mu #1523: Crash Report

soarek, 2021-05-01:
Hi, can someone help me to understand where is the issue and how to fix it?
Thanks.

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: pl_PL
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero
[mu.log](https://github.com/mu-editor/mu/files/6409243/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/6409245/mu.log)

```

Here the full log:

2021-05-01 09:48:09,428 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-01 09:48:09,429 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-GHQJGM1', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 12, GenuineIntel')
2021-05-01 09:48:09,429 - root:267(run) INFO: Python path: ['C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\soare\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-01 09:48:09,429 - root:268(run) INFO: Language code: pl_PL
2021-05-01 09:48:09,429 - mu.settings:220(load) WARNING: No settings file found at C:\Users\soare\AppData\Local\python\mu\settings.json; skipping
2021-05-01 09:48:09,578 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-01 09:48:09,578 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-01 09:48:09,578 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\soare\AppData\Local\python\mu\mu_venv-38-20210430-083410
2021-05-01 09:48:09,578 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210430-083410
2021-05-01 09:48:11,049 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-01 09:48:11,049 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\soare\AppData\Local\python\mu\mu_venv-38-20210430-083410
2021-05-01 09:48:11,049 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\soare\AppData\Local\python\mu\mu_venv-38-20210430-083410\scripts\python.exe exists
2021-05-01 09:48:27,395 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: Requirement already satisfied: pip in c:\users\soare\appdata\local\python\mu\mu_venv-38-20210430-083410\lib\site-packages (21.0.1)
Collecting pip
  Downloading pip-21.1.1-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0.1
    Uninstalling pip-21.0.1:
      Successfully uninstalled pip-21.0.1
Successfully installed pip-21.1.1
2021-05-01 09:48:27,395 - mu.virtual_environment:669(upgrade_pip) INFO: Upgraded pip
2021-05-01 09:48:27,395 - mu.virtual_environment:711(install_baseline_packages) INFO: Installing baseline packages.
2021-05-01 09:48:27,395 - mu.virtual_environment:712(install_baseline_packages) INFO: C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-05-01 09:48:27,429 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-05-01 09:48:29,389 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\soare\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\soare\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl

Processing c:\users\soare\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl

Processing c:\users\soare\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl

Processing c:\users\soare\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl

Processing c:\users\soare\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl

[... 2209 lines cut ...]
    The user requested cryptography 3.4.6 (from C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\cryptography-3.4.6-cp36-abi3-win_amd64.whl)

    The user requested cryptography 3.4.7 (from C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl)

To fix this you could try to:

1. loosen the range of package versions you've specified

2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies

2021-05-01 10:22:13,682 - mu.virtual_environment:607(install_jupyter_kernel) INFO: Installing Jupyter Kernel: "Python/Mu (mu_venv-38-20210501-102209)"
2021-05-01 10:22:13,878 - mu.virtual_environment:456(ensure_and_create) ERROR: Unable to create a working virtual environment.
2021-05-01 10:22:20,898 - root:174(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\soare\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 459, in ensure_and_create
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Unable to create a working virtual environment.
2021-05-01 10:22:21,068 - mu.settings:169(save) DEBUG: Saving to C:\Users\soare\AppData\Local\python\mu\session.json
2021-05-01 10:22:21,068 - mu.settings:169(save) DEBUG: Saving to C:\Users\soare\AppData\Local\python\mu\venv.json

tjguk, 2021-05-03:
Thanks for reporting this, @soarek  -- can I confirm that you installed from the downloadable installer? (I suspect you did, going by the paths, but I'm just checking...)

tjguk, 2021-05-03:
Also -- was this an upgrade from a previous beta, or from some other installation of Mu? Or was this the first time you installed?

carlosperate, 2021-08-04:
Thanks for the report @soarek!
I'll close this specific issue as a being covered in https://github.com/mu-editor/mu/issues/1421.

If you could have a look at the release mentioned in this issue response and try the new Mu installer that would be really useful for us to confirm that it is fixed: https://github.com/mu-editor/mu/issues/1421#issuecomment-892776083

Thanks!

closed 2021-08-04 by carlosperate