mu-editor/mu #1524: pi pico mu circiutpython

jimbo95-beep, 2021-05-01:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

jimbo95-beep, 2021-05-01:
no module neopixel  or no board 
I have been trying to run  neopixel strip addressable leds 
using circuitpython  run a code rainbow.py adafruit library 
but comes up with no module neopixel or no board module ?
I have add uptodate circuitpython file.
- any help please

dsssssssss9, 2021-05-01:
Can you let us see your code please so we can have an better idea of how to help you

ajs256, 2021-05-01:
Are you sure Mu is in CircuitPython mode? Click the button at the top left with Mu's logo and choose "CircuitPython" - the one with the purple snake.

jimbo95-beep, 2021-05-10:
I was trying to get addressable LEDs to on adafruit neopixel library programme rainbow
I was using raspberry pi pico board.

I install uf2 file and the added the circuitpython uf2 file 
As I  was using a circuit python code.
I had connected my wires to the board one on ground and o e 5v
I also had extra power as it was a strip of LEDs 300 approx
I had the other work flash on broad led. 
Test programme. 
And a senior buzzer programme turns buzzer if the sensor is triped 
They both worked.

Skip to content

Sign up

adafruit/Adafruit_Learning_System_Guides

 Notifications

 Star466

 Fork459

Code

Issues13

Pull requests14

Actions

Security

Insights

 master 

Adafruit_Learning_System_Guides/CircuitPython_Essentials/CircuitPython_NeoPixel_RGBW.py / Jump to 

Go to file

￼

kattni Updating Essentials guide to include QT Py.

Latest commit 3e07b2f on Dec 16, 2020 History

 2 contributors￼￼

70 lines (57 sloc)  1.7 KB

RawBlame

 

"""CircuitPython Essentials NeoPixel RGBW example"""import timeimport boardimport neopixel pixel_pin = board.A1num_pixels = 8 pixels = neopixel.NeoPixel(pixel_pin, num_pixels, brightness=0.3, auto_write=False, pixel_order=(1, 0, 2, 3)) def wheel(pos): # Input a value 0 to 255 to get a color value. # The colours are a transition r - g - b - back to r. if pos < 0 or pos > 255: return (0, 0, 0, 0) if pos < 85: return (255 - pos * 3, pos * 3, 0, 0) if pos < 170: pos -= 85 return (0, 255 - pos * 3, pos * 3, 0) pos -= 170 return (pos * 3, 0, 255 - pos * 3, 0) def color_chase(color, wait): for i in range(num_pixels): pixels[i] = color time.sleep(wait) pixels.show() time.sleep(0.5) def rainbow_cycle(wait): for j in range(255): for i in range(num_pixels): rc_index = (i * 256 // num_pixels) + j pixels[i] = wheel(rc_index & 255) pixels.show() time.sleep(wait) RED = (255, 0, 0, 0)YELLOW = (255, 150, 0, 0)GREEN = (0, 255, 0, 0)CYAN = (0, 255, 255, 0)BLUE = (0, 0, 255, 0)PURPLE = (180, 0, 255, 0) while True: pixels.fill(RED) pixels.show() # Increase or decrease to change the speed of the solid color change. time.sleep(1) pixels.fill(GREEN) pixels.show() time.sleep(1) pixels.fill(BLUE) pixels.

When running this code l get no module found board or neopixel 

What do need to install add to the library so it recognize board and neopixel?

ajs256, 2021-05-10:
The `board` module is built in to CircuitPython, but the `neopixel` one is not. Grab it from the bundle, [here](https://circuitpython.org/libraries). 

As for the fact that the board module wasn't found, that is not normal. Are you sure Mu is in CircuitPython mode? Click the button in the top left corner with Mu's logo, and choose CircuitPython.