mu-editor/mu #1525: Crash on start when connected to Adafruit RP2040, RP Pico, Microbit

Ardo9602, 2021-05-01:
Adafruit RP2040 with Circuitpython 6.2.0 beta4 and release
Microbit with Micropython

Editor starts fine with no board connected, crashes with any of these connected, connecting before or after program start makes no difference.

Here's the log:
2021-05-01 22:58:51,442 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-01 22:58:51,442 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-14PAKMJ', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 158 Stepping 10, GenuineIntel')
2021-05-01 22:58:51,442 - root:267(run) INFO: Python path: ['C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Admiq\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-01 22:58:51,457 - root:268(run) INFO: Language code: en_US
2021-05-01 22:58:51,457 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Admiq\AppData\Local\python\mu\settings.json; skipping
2021-05-01 22:58:51,511 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-01 22:58:51,645 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-01 22:58:51,645 - mu.virtual_environment:542(ensure_path) INFO: Virtual Environment found at: C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210430-215010
2021-05-01 22:58:51,654 - mu.virtual_environment:555(ensure_interpreter) INFO: Interpreter found at: C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210430-215010\scripts\python.exe
2021-05-01 22:58:51,729 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 38
2021-05-01 22:58:51,729 - mu.virtual_environment:585(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-05-01 22:58:51,730 - mu.virtual_environment:611(ensure_pip) INFO: Pip found at: C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210430-215010\scripts\pip.exe
2021-05-01 22:58:51,730 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-05-01 22:58:51,805 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-01 22:58:51,805 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: flask
2021-05-01 22:58:52,148 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-01 22:58:52,148 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-05-01 22:58:52,244 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-01 22:58:52,245 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-05-01 22:58:52,393 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-01 22:58:52,393 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: adafruit_board_toolkit
2021-05-01 22:58:52,487 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-01 22:58:52,487 - mu.virtual_environment:496(ensure_and_create) INFO: Valid virtual environment found at C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210430-215010
2021-05-01 22:58:52,487 - mu.settings:169(save) DEBUG: Saving to C:\Users\Admiq\AppData\Local\python\mu\venv.json
2021-05-01 22:58:52,489 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x000001C8C87FA5E0>; Splash Handler; <SplashLogHandler (INFO)>
2021-05-01 22:58:52,585 - mu.logic:758(__init__) INFO: Setting up editor.
2021-05-01 22:58:52,585 - mu.logic:778(__init__) INFO: Log directory: C:\Users\Admiq\AppData\Local\python\mu\Logs
2021-05-01 22:58:52,585 - mu.logic:779(__init__) INFO: Data directory: C:\Users\Admiq\AppData\Local\python\mu
2021-05-01 22:58:52,585 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-05-01 22:58:52,586 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Admiq\mu_code from settings file
2021-05-01 22:58:52,636 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Admiq\mu_code from settings file
2021-05-01 22:58:52,636 - mu.logic:1521(change_mode) INFO: Workspace directory: C:\Users\Admiq\mu_code
2021-05-01 22:58:52,636 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\Admiq\AppData\Local\python\mu\session.json>
2021-05-01 22:58:52,643 - mu.logic:889(restore_session) INFO: Loaded files.
2021-05-01 22:58:52,643 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-05-01 22:58:52,643 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
2021-05-01 22:58:52,675 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Admiq\mu_code from settings file
2021-05-01 22:58:52,675 - mu.logic:1521(change_mode) INFO: Workspace directory: C:\Users\Admiq\mu_code
2021-05-01 22:58:52,872 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-05-01 22:58:53,593 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 711, in check_usb
    detected = mode.find_devices(with_logging=False)
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
    device = self.compatible_board(port)
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 286, in compatible_board
    for comport in circuitpython_serial.repl_comports():
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 61, in repl_comports
    port for port in comports() if port.interface.startswith("CircuitPython CDC ")
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 46, in comports
    ports = _list_ports_windows.comports()
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 458, in comports
    return list(iterate_comports())
  File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'
2021-05-01 22:58:53,744 - mu.settings:169(save) DEBUG: Saving to C:\Users\Admiq\AppData\Local\python\mu\session.json
2021-05-01 22:58:53,745 - mu.settings:169(save) DEBUG: Saving to C:\Users\Admiq\AppData\Local\python\mu\venv.json
2021-05-01 23:04:32,783 - root:265(run) INFO: 

[... 170 lines cut ...]

pywin32                300

pyzmq                  22.0.3

qtconsole              4.7.4

QtPy                   1.9.0

reedsolo               1.5.4

setuptools             56.0.0

six                    1.15.0

tornado                6.1

traitlets              5.0.5

wcwidth                0.2.5

Werkzeug               1.0.1

wheel                  0.36.2
2021-05-01 23:05:00,398 - mu.virtual_environment:797(installed_packages) INFO: []

dhalbert, 2021-05-02:
This is some kind of version problem The error is
```
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit_list_ports_windows.py", line 342, in iterate_comports
info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'
```
The keyword argument `skip_link_detection` was added in pyserial 3.5, but is not present in pyserial 3.4. Mu is now requiring pyserial 3.5. According to the log, you are using pyserial 3.5, but that bug would only happen true if Mu somehow loaded pyserial 3.4 instead.

Mu developers, do you understand how this could happen?

ntoll, 2021-05-02:
Calling @dhalbert - this seems to be a problem in the `adafruit_board_toolkit`. I'll create an issue in your repos and link it to this one just so we can join the dots. The full log is copied in the OP, but this is the stack trace with where things go boom. If there's anything we can do to help, please don't hesitate to ask..! :+1: :rocket: 

```
2021-05-01 23:04:35,139 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 711, in check_usb
detected = mode.find_devices(with_logging=False)
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
device = self.compatible_board(port)
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 286, in compatible_board
for comport in circuitpython_serial.repl_comports():
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 61, in repl_comports
port for port in comports() if port.interface.startswith("CircuitPython CDC ")
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 46, in comports
ports = _list_ports_windows.comports()
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit_list_ports_windows.py", line 458, in comports
return list(iterate_comports())
File "C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit_list_ports_windows.py", line 342, in iterate_comports
info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: init() got an unexpected keyword argument 'skip_link_detection'
```

ntoll, 2021-05-02:
@dhalbert apologies for not spotting your comment (I was looking at a version of the page I opened before you added your comment). :facepalm:

Mu *should* require PySerial 3.5. @Ardo9602 how did you install Mu..? It's not clear what's going on given that you appear to show you're using PySerial 3.5. The only thing I can think of is if there is somehow an old `.pyc` file (for PySerial < 3.5) in the virtualenv created by a previous version of Mu.

What you could try is deleting the virtualenv directory.... from the logs you should find it here: `C:\Users\Admiq\AppData\Local\python\mu\`

In fact, to create a "clean install", you could try deleting all the files in that directory and then starting up Mu. Had you installed a previous Mu Beta release before trying Beta 4..?

Ardo9602, 2021-05-02:
I wasn't even aware of Mu's existence before last friday....; this happened on my very first install, but I'll give it a try, deleting ...\python\mu alltogether and un-/reinstall

Ardo9602, 2021-05-02:
Same thing, crash

ntoll, 2021-05-02:
Could you paste in the new logs please..?

Ardo9602, 2021-05-02:
2021-05-02 11:58:00,886 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-02 11:58:00,886 - root:266(run) INFO: uname_result(system='Windows', node='DESKTOP-14PAKMJ', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 158 Stepping 10, GenuineIntel')
2021-05-02 11:58:00,886 - root:267(run) INFO: Python path: ['C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Admiq\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Admiq\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-02 11:58:00,886 - root:268(run) INFO: Language code: en_US
2021-05-02 11:58:00,886 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Admiq\AppData\Local\python\mu\settings.json; skipping
2021-05-02 11:58:00,886 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Admiq\AppData\Local\python\mu\session.json; skipping
2021-05-02 11:58:01,447 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-02 11:58:01,779 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-02 11:58:01,779 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210502-115747
2021-05-02 11:58:01,795 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210502-115747
2021-05-02 11:58:08,279 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-02 11:58:08,279 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210502-115747
2021-05-02 11:58:08,279 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\Admiq\AppData\Local\python\mu\mu_venv-38-20210502-115747\scripts\python.exe exists
2021-05-02 11:58:15,614 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: Requirement already satisfied: pip in c:\users\admiq\appdata\local\python\mu\mu_venv-38-20210502-115747\lib\site-packages (21.0.1)
Collecting pip
  Using cached pip-21.1.1-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0.1
    Uninstalling pip-21.0.1:
      Successfully uninstalled pip-21.0.1
Successfully installed pip-21.1.1
2021-05-02 11:58:15,614 - mu.virtual_environment:669(upgrade_pip) INFO: Upgraded pip
2021-05-02 11:58:15,614 - mu.virtual_environment:711(install_baseline_packages) INFO: Installing baseline packages.
2021-05-02 11:58:15,614 - mu.virtual_environment:712(install_baseline_packages) INFO: C:\Users\Admiq\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-05-02 11:58:15,614 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-05-02 11:58:55,550 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cryptography-3.4.7-cp36-abi3-win_amd64.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\decorator-5.0.7-py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\ecdsa-0.16.1-py2.py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\esptool-3.0-py3-none-any.whl

Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\flask-1.1.2-py2.py3-none-any.whl

[... cut ...]
Processing c:\users\admiq\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\markupsafe-1.1.1-cp38-cp38-win_amd64.whl

[... 424 lines cut ...]

pywin32                300

pyzmq                  22.0.3

qtconsole              4.7.4

QtPy                   1.9.0

reedsolo               1.5.4

setuptools             56.0.0

six                    1.15.0

tornado                6.1

traitlets              5.0.5

wcwidth                0.2.5

Werkzeug               1.0.1

wheel                  0.36.2
2021-05-02 13:30:22,901 - mu.virtual_environment:797(installed_packages) INFO: []

carlosperate, 2021-08-10:
Hi @Ardo9602,

Could you copy and paste this code in a new file, run it in Mu in Python3 mode, and paste here the result?

This should give us a bit more info about how come your Mu installation has pyserial v3.5 in the filesystem, but somehow the version running in Mu might be 3.4 or older.

carlosperate, 2022-06-03:
Hi @Ardo9602,

Would you be able to try windows installer from this page? https://github.com/mu-editor/mu/actions/runs/2414316036

You can download it from the Artifacts section at the bottom of the page (make sure you are logged in GitHub for the download link to work):
[image]

I think the updates in that test installer could resolve the issue you have. It would be really useful for us if you could test it, thanks!