mu-editor/mu #1526: Crash Report

Elikyals, 2021-05-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Yals\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Yals\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Yals\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Yals\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Yals\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-05-05:
Thanks for the report @Elikyals!
I'll reopen this issue as it might be a problem we haven't encountered yet.
Would you be able to attach the Mu log here as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

Elikyals, 2021-05-05:
@carlosperate I did not encounter the problem any longer after I uninstalled and reinstalled. When installing the latest version (beta 4) I did not uninstall the old version (beta 3), I think that's what cause the problem.

carlosperate, 2021-05-05:
Thanks @Elikyals, I think you are right, in that case it looks like issue https://github.com/mu-editor/mu/issues/1421.

We can close this ticket as we'll track progress for that fix over ther.

Thanks again!

closed 2021-05-05 by carlosperate

Elikyals, 2021-05-05:
@carlosperate Thanks for the feedback