mu-editor/mu #1527: Crash Report

pradputtur, 2021-05-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_IN
Platform: Windows 7 6.1.7601 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Ravi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Ravi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Ravi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Ravi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Ravi\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

pradputtur, 2021-05-03:
Hi- I am facing an issue while opening mu tool- this is the first time I am using python and mu

tjguk, 2021-05-03:
Sorry about that @pradputtur  -- would you be able to upload logs from `%LOCALAPPDATA%\python\mu\logs` please?

tjguk, 2021-05-03:
(BTW I note that you're running on Windows7 -- and I don't think we have a test machine running that version of Windows, so there may be some issues there...)

carlosperate, 2021-06-04:
Hi @pradputtur, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\Ravi\AppData\Local\python\mu\logs`.

carlosperate, 2021-08-10:
Hi @pradputtur,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Thanks!

carlosperate, 2021-09-21:
@pradputtur have you been able to test beta.5? There has been some fixes in this area that could have resolved the problem you were experiencing: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate