mu-editor/mu #1528: Crash Report

falcen99, 2021-05-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: cs_CZ
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Pejsek\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Pejsek\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Pejsek\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Pejsek\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Pejsek\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

tjguk, 2021-05-03:
Thanks @falcen99  -- would you be able to upload the log files, please, from `%LOCALAPPDATA%\python\mu\logs`?

carlosperate, 2021-06-04:
Hi @falcen99, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\Pejsek\AppData\Local\python\mu\logs`.

carlosperate, 2021-08-10:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-08-10 by carlosperate