mu-editor/mu #1531: Run script do a soft reboot of Witty Cloud (ESP8266) with MicroPython v1.15 (stable)

4SiB, 2021-05-03:
If you are reporting a bug, we would like to know:

* What you were trying to do,
I write a test.py with one line: x = 1
I press Run button and see "ESP MicroPython REPL" errors about soft reboot.

* What steps you took to make this happen,
I try RUN script.

* What you expected to happen,
See results below and in future use Plotter with LDR but now REPL not with Mu

* What actually happened,
This: 
[image]

* Why this difference is problematic (it may not be a bug!),
If this is not bug then MU should show a information what to do to MU start work properly.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Mu 1.1.0beta4 - latest from 10 minutes ago
Windows 10.
By COM I can see and use REPL when I not use MU of course.
By ws://IP:8266 I can see and use WebREPL when I not use MU of course.

Please remember to attach a **copy of the full log files for Mu**.
[image]

2021-05-03 22:04:22,205 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-05-03 22:04:22,206 - root:266(run) INFO: uname_result(system='Windows', node='SIB-E5480', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 9, GenuineIntel')
2021-05-03 22:04:22,206 - root:267(run) INFO: Python path: ['C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-05-03 22:04:22,213 - root:268(run) INFO: Language code: pl_PL
2021-05-03 22:04:22,214 - mu.settings:220(load) WARNING: No settings file found at C:\Users\admin\AppData\Local\python\mu\settings.json; skipping
2021-05-03 22:04:22,214 - mu.settings:220(load) WARNING: No settings file found at C:\Users\admin\AppData\Local\python\mu\session.json; skipping
2021-05-03 22:04:22,296 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-05-03 22:04:22,979 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-05-03 22:04:22,979 - mu.virtual_environment:633(create_venv) INFO: Creating virtualenv: C:\Users\admin\AppData\Local\python\mu\mu_venv-38-20210503-220417
2021-05-03 22:04:22,979 - mu.virtual_environment:634(create_venv) INFO: Virtualenv name: mu_venv-38-20210503-220417
2021-05-03 22:04:38,946 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-05-03 22:04:38,946 - mu.virtual_environment:648(create_venv) INFO: Created virtual environment using C:\Users\admin\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\admin\AppData\Local\python\mu\mu_venv-38-20210503-220417
2021-05-03 22:04:38,968 - mu.virtual_environment:660(upgrade_pip) DEBUG: About to upgrade pip; interpreter C:\Users\admin\AppData\Local\python\mu\mu_venv-38-20210503-220417\scripts\python.exe exists
2021-05-03 22:04:55,076 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: Requirement already satisfied: pip in c:\users\admin\appdata\local\python\mu\mu_venv-38-20210503-220417\lib\site-packages (21.0.1)
Collecting pip
  Using cached pip-21.1.1-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0.1
    Uninstalling pip-21.0.1:
      Successfully uninstalled pip-21.0.1
Successfully installed pip-21.1.1
2021-05-03 22:04:55,076 - mu.virtual_environment:669(upgrade_pip) INFO: Upgraded pip
2021-05-03 22:04:55,077 - mu.virtual_environment:711(install_baseline_packages) INFO: Installing baseline packages.
2021-05-03 22:04:55,078 - mu.virtual_environment:712(install_baseline_packages) INFO: C:\Users\admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels exists
2021-05-03 22:04:55,107 - mu.virtual_environment:743(install_baseline_packages) INFO: About to install from wheels
2021-05-03 22:05:45,485 - mu.virtual_environment:193(run) DEBUG: Process output: Processing c:\users\admin\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\adafruit_board_toolkit-1.0.1-py3-none-any.whl

Processing c:\users\admin\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\backcall-0.2.0-py2.py3-none-any.whl

Processing c:\users\admin\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\bitstring-3.1.7-py3-none-any.whl

Processing c:\users\admin\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\cffi-1.14.5-cp38-cp38-win_amd64.whl

Processing c:\users\admin\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\click-7.1.2-py2.py3-none-any.whl

Processing c:\users\admin\appdata\local\programs\mu editor\python\lib\site-packages\mu\wheels\colorama-0.4.4-py2.py3-none-any.whl
[... 987 lines cut ...]

pywin32                300

pyzmq                  22.0.3

qtconsole              4.7.4

QtPy                   1.9.0

reedsolo               1.5.4

setuptools             56.0.0

six                    1.15.0

tornado                6.1

traitlets              5.0.5

wcwidth                0.2.5

Werkzeug               1.0.1

wheel                  0.36.2
2021-05-03 22:30:21,078 - mu.virtual_environment:797(installed_packages) INFO: []

carlosperate, 2021-05-04:
Hi @4SiB,

This looks like the expected behaviour. Everytime the scipt is run it will try to enter the REPL, force a soft reset, reset the device (which in turn runs the boot.py script) and then run your code from main.py.

The output you currently see is from all the steps mentioned above. As your code is not doing anything that would be reflected in the serial output then this is all you see. Adding other code, for example `print("Hello world)"`, should show up in that panel.

4SiB, 2021-05-04:
I really want to be true but even with print("Hello world)" the situation is the same.
[image]

You see, the output not exist, REPL show my WebREPL and this
`
>OKTraceback (most recent call last):
  File "<stdin>", line 4
SyntaxError: invalid syntax
>
`
----------

I flash again new v1.15 esp8266... and this still not work properly.

[image]

4SiB, 2021-05-04:
I do more, I re-flash version 1.15, 1.14, 1.13, 1.12 and all them have the same/similar error with a SyntaxError: invalid syntax.

Logs from 1.12:

> 2021-05-04 19:41:21,328 - mu.modes.esp:175(run) INFO: Running script.
> 2021-05-04 19:41:21,328 - mu.modes.base:112(open) INFO: Connecting to REPL on port: COM5
> 2021-05-04 19:41:21,333 - mu.modes.base:130(open) INFO: Connected to REPL on port: COM5
> 2021-05-04 19:41:21,339 - mu.modes.base:503(add_repl) INFO: Started REPL on port: COM5
> 2021-05-04 19:41:21,339 - mu.modes.base:474(toggle_repl) INFO: Toggle REPL on.
> 2021-05-04 19:41:21,339 - mu.modes.base:188(send_commands) INFO: [b'\x03', b'\x03', b'\x01', b'\x04', b'\x03', b'\x03', b'print("\\n");', b'print("Hello world)"\r', b'\r', b'\r', b'\x04', b'\x02']
> 2021-05-04 19:41:21,339 - mu.modes.base:162(execute) INFO: Sending command b'\x03'
> 2021-05-04 19:41:21,344 - mu.modes.base:162(execute) INFO: Sending command b'\x03'
> 2021-05-04 19:41:21,349 - mu.modes.base:162(execute) INFO: Sending command b'\x01'
> 2021-05-04 19:41:21,357 - mu.modes.base:162(execute) INFO: Sending command b'\x04'
> 2021-05-04 19:41:21,361 - mu.modes.base:162(execute) INFO: Sending command b'\x03'
> 2021-05-04 19:41:21,363 - mu.modes.base:162(execute) INFO: Sending command b'\x03'
> 2021-05-04 19:41:21,366 - mu.modes.base:162(execute) INFO: Sending command b'print("\\n");'
> 2021-05-04 19:41:21,369 - mu.modes.base:162(execute) INFO: Sending command b'print("Hello world)"\r'
> 2021-05-04 19:41:21,371 - mu.modes.base:162(execute) INFO: Sending command b'\r'
> 2021-05-04 19:41:21,374 - mu.modes.base:162(execute) INFO: Sending command b'\r'
> 2021-05-04 19:41:21,376 - mu.modes.base:162(execute) INFO: Sending command b'\x04'
> 2021-05-04 19:41:21,378 - mu.modes.base:162(execute) INFO: Sending command b'\x02'
> 2021-05-04 19:41:25,328 - mu.logic:1394(show_admin) INFO: Showing admin with logs from C:\Users\admin\AppData\Local\python\mu\Logs\mu.log
> 2021-05-04 19:41:25,328 - mu.virtual_environment:788(installed_packages) INFO: Discovering installed third party modules in venv.
> 2021-05-04 19:41:27,038 - mu.virtual_environment:193(run) DEBUG: Process output: Package                Version
> 
> ---------------------- -------
> 
> adafruit-board-toolkit 1.0.1
> 
> backcall               0.2.0
> 
> bitstring              3.1.7
> 
> cffi                   1.14.5
> 
> click                  7.1.2
> 
> colorama               0.4.4
> 
> cryptography           3.4.7
> 
> decorator              5.0.7
> 
> ecdsa                  0.16.1
> 
> esptool                3.0
> 
> Flask                  1.1.2
> 
> ipykernel              5.5.3
> 
> ipython                7.22.0
> 
> ipython-genutils       0.2.0
> 
> itsdangerous           1.1.0
> 
> jedi                   0.18.0
> 
> Jinja2                 2.11.3
> 
> jupyter-client         6.1.12
> 
> jupyter-core           4.7.1
> 
> MarkupSafe             1.1.1
> 
> numpy                  1.20.2
> 
> parso                  0.8.2
> 
> pgzero                 1.2.1
> 
> pickleshare            0.7.5
> 
> pip                    21.1.1
> 
> prompt-toolkit         3.0.18
> 
> pycparser              2.20
> 
> pygame                 2.0.1
> 
> Pygments               2.8.1
> 
> pyserial               3.5
> 
> python-dateutil        2.8.1
> 
> pywin32                300
> 
> pyzmq                  22.0.3
> 
> qtconsole              4.7.4
> 
> QtPy                   1.9.0
> 
> reedsolo               1.5.4
> 
> setuptools             56.0.0
> 
> six                    1.15.0
> 
> tornado                6.1
> 
> traitlets              5.0.5
> 
> wcwidth                0.2.5
> 
> Werkzeug               1.0.1
> 
> wheel                  0.36.2
> 2021-05-04 19:41:27,038 - mu.virtual_environment:797(installed_packages) INFO: []

carlosperate, 2021-05-04:
Hi @4SiB,

That's because there is an error in your code, the bracket and quote marks are reversed: `print("Hello world)"` should be `print("Hello world")`

4SiB, 2021-05-04:
> Hi @4SiB,
> 
> for example `print("Hello world)"`, should show up in that panel.

You do error and I copy-paste it without thinking.
I discover it too :).

And now yes, it's work and my final target it's done !.
Problem solve !.

Plotter works with MicroPython 1.15 with this code:
```
import time
import urandom

def randint(min, max):
    span = max - min + 1
    div = 0x3fffffff // span
    offset = urandom.getrandbits(30) // div
    val = min + offset
    return val

while True:
    time.sleep(0.05)
    print((randint(0, 100), randint(-100, 0), randint(-50, 50)))

```

[image]

closed 2021-05-04 by 4SiB

carlosperate, 2021-05-04:
You are 100% right, that was my bad 😅
Glad you got it working!