mu-editor/mu #1532: Crash Report

rumeysaarslan, 2021-05-04:
[mu.log](https://github.com/mu-editor/mu/files/6421371/mu.log)

**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\rumey\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\rumey\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\rumey\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\rumey\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\rumey\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

carlosperate, 2021-05-04:
@tjguk looks like there can be issues upgrading the venv pip (this is might assumption from a quick look at the error, might be worth having a better look in case this is a different problem).

One of the advantages of using `virtualenv` is that we can let it manage its pip version, and we can count as the version being modern if we set up a min version of virtualenv in setup.py.
https://virtualenv.pypa.io/en/latest/user_guide.html#seeders

Of course, if we have issues installing pip we are likely going to have issues installing other packages in the venv. But taking in consideration how complex this system can be, reducing the number of things that can break can still be valuable.

carlosperate, 2021-05-04:
Also, I guess upgrading pip in the venv requires an internet connection?
In this case to support installing Mu offline and running Mu offline (specially on the first run) we would need to check for this specific "no internet connection" error and still try to install all the wheels if that step fails.

tjguk, 2021-05-04:
Good point re internet connection @carlosperate  -- but are you seeing that from this user's log? I can't see anything which points to an internet connection error.

carlosperate, 2021-05-04:
Oh no, the internet connection was a though I had after I wrote the previous message.

From the log it looks like the venv contains a slightly older pip (21.0) and when trying to uninstall it to install the newer version (21.1) it fails at some point.

```
 DEBUG: Process returned 3221225786; output: Requirement already satisfied: pip in c:\users\rumey\appdata\local\python\mu\mu_venv-38-20210504-161913\lib\site-packages (21.0.1)
Collecting pip
  Using cached pip-21.1.1-py3-none-any.whl (1.5 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.0.1
    Uninstalling pip-21.0.1:
      Successfully uninstalled pip-21.0.1
2021-05-04 16:19:56,239 - mu.virtual_environment:468(quarantine_failed_venv) INFO: Quarantined C:\Users\rumey\AppData\Local\python\mu\mu_venv-38-20210504-161913 as C:\Users\rumey\AppData\Local\python\mu\mu_venv-38-20210504-161913.FAILED
2021-05-04 16:19:56,245 - mu.virtual_environment:410(relocate) DEBUG: Virtual environment set up mu_venv-38-20210504-161956 at C:\Users\rumey\AppData\Local\python\mu\mu_venv-38-20210504-161956
```

Looks like maybe the log is capturing the stdout but not the stderr? From a terminal I'd expect the problem to be printed after the `Successfully uninstalled pip-21.0.1` line.

tjguk, 2021-05-06:
(Logs from #1538 showing problems upgrading pip via a proxy)
[mu.log](https://github.com/mu-editor/mu/files/6433547/mu.log)

tjguk, 2021-05-06:
(Logs from #1541)

[mu-1541.log](https://github.com/mu-editor/mu/files/6435047/mu-1541.log)

carlosperate, 2021-05-11:
No logs, but same issue in https://github.com/mu-editor/mu/issues/1557.

carlosperate, 2021-05-11:
Merging https://github.com/mu-editor/mu/issues/1555 here:

> [mu.log](https://github.com/mu-editor/mu/files/6451057/mu.log)
>
> Version: 1.1.0.beta.4
> Locale: fr_CH
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\ENVY 17-r185\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\ENVY 17-r185\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\ENVY 17-r185\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\ENVY 17-r185\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\ENVY 17-r185\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> 
> ```

carlosperate, 2021-06-14:
Merging https://github.com/mu-editor/mu/issues/1637 here:

> [mu.log](https://github.com/mu-editor/mu/files/6643322/mu.log)
> [mu.log](https://github.com/mu-editor/mu/files/6643324/mu.log)
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Jerry Hatchett\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Jerry Hatchett\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Jerry Hatchett\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\Jerry Hatchett\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\Jerry Hatchett\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-06-14:
Merging https://github.com/mu-editor/mu/issues/1639 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_IN
> Platform: Windows 10 10.0.18362 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\muthuswamy\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\muthuswamy\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\muthuswamy\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\muthuswamy\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\muthuswamy\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-06-14:
Merging https://github.com/mu-editor/mu/issues/1609 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\Rick 17\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-06-14:
Merging https://github.com/mu-editor/mu/issues/1629 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_US
> Platform: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Thu May 6 00:48:39 PDT 2021; root:xnu-6153.141.33~1/RELEASE_X86_64 x86_64
> 
> ```
> Traceback (most recent call last):
>   File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 161, in run
>     raise ex
>   File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "/private/var/folders/5y/g45lwy053sbfwmmkr9sw39wc0000gn/T/AppTranslocation/9FB9FDA5-AA07-48BD-8A19-A125E1485D33/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-06-14:
Mu no longer tries to update the venv pip version via PR https://github.com/mu-editor/mu/pull/1624, it now relays on the `virtualenv` module having a moderately modern version of pip vendored via PR https://github.com/mu-editor/mu/pull/1573.

So this issue should be resolved in the next beta release.

carlosperate, 2021-06-15:
Merging https://github.com/mu-editor/mu/issues/1642 here:

> ```
> Traceback (most recent call last):
>   File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\Pieter Viljoen\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```
> 
> log file: [mu.log](https://github.com/mu-editor/mu-editor.github.io/files/6470001/mu.log)

carlosperate, 2021-08-04:
@rumeysaarslan @pviljoen2 @cbrown56 @RE2222 @Muthuswamy18 @ChopOMatic @luchskeudenn @zaneze @AbouTheGreat @MCEITsupport

Hi all!

The issues you were having with Mu should have been resolved with the beta.5 release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Would you be able to try the installer from that release and check if it is working now?

Thanks everyone for your help testing the Mu Editor!

carlosperate, 2021-08-06:
Merging https://github.com/mu-editor/mu/issues/1540 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: fr_FR
> Platform: Windows 10 10.0.18362 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-08-06:
Merging https://github.com/mu-editor/mu/issues/1648 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: zh_TW
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\dpshop Buyer2\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\dpshop Buyer2\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\dpshop Buyer2\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\dpshop Buyer2\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\dpshop Buyer2\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-08-06:
Merging https://github.com/mu-editor/mu/issues/1652 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_GB
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-08-06:
Merging https://github.com/mu-editor/mu/issues/1667 here:

> ### What were you trying to do?
> 
> install Mu Editor
> ### What steps did you take to trigger the issue?
> 
> I installed Mu Editor.. I ran it after the install was done and then this
> ### What did you expect to happen?
> 
> I would b able to use the program
> ### What actually happened?
> 
> this Crash report happened
> ### Operating System Version
> 
> Windows 10
> ### Mu Version
> 
> 1.10

> ### Editor Log
> 
> Traceback (most recent call last):
> File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
> raise ex
> File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
> venv.ensure_and_create(self.display_text)
> File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
> self.create()
> File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
> self.upgrade_pip()
> File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
> raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

carlosperate, 2021-08-06:
Merging https://github.com/mu-editor/mu/issues/1675 here:

> **REMEMBER TO ATTACH YOUR LOG FILE**
> 
> Version: 1.1.0.beta.4
> Locale: en_CA
> Platform: Windows 10 10.0.19041 AMD64
> 
> ```
> Traceback (most recent call last):
>   File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
>     raise ex
>   File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
>     venv.ensure_and_create(self.display_text)
>   File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
>     self.create()
>   File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
>     self.upgrade_pip()
>   File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
>     raise VirtualEnvironmentCreateError("Unable to upgrade pip")
> mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
> ```

carlosperate, 2021-08-06:
@brunel-bit @Jason0930 @RogerArtemis @DLadyE @Roshelys

Hi all!

The issues you were having with Mu should have been resolved with the beta.5 release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

Would you be able to try the installer from that release and let us know if it is working now?

Thanks everyone for your help testing the Mu Editor!

carlosperate, 2021-09-22:
We haven't received any reports about this issue since beta.5 was released, so I'll consider this resolved.
Thanks everybody!

closed 2021-09-22 by carlosperate