mu-editor/mu #1535: Crash Report

JojoRushing, 2021-05-05:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: zh_TW
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 344, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\a1536\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] ç¶æªæ¡å·²å­å¨æï¼ç¡æ³å»ºç«è©²æªæ¡ã: 'C:\\Users\\a1536\\mu_code'

```

carlosperate, 2021-05-05:
This is related to https://github.com/mu-editor/mu/issues/1506, where we first check if a directory/file exists before creating it, and then it fails in the directory creation step saying that it already exists:
https://github.com/mu-editor/mu/blob/2b20da24e5219002fe19781509fa8665a9842464/mu/logic.py#L796-L798

We have very similar code in multiple places, so would be good to figure out what might be happening.

carlosperate, 2021-06-04:
Hi @JojoRushing, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\a1536\AppData\Local\python\mu\logs`.

carlosperate, 2021-09-27:
Closing due to lack of activity, the issue in itself is also tracked in https://github.com/mu-editor/mu/issues/1755.

closed 2021-09-27 by carlosperate