mu-editor/mu #1536: unreliable flashing of microbit v2 (521 error)

Maxelon, 2021-05-05:
Had this issue with beta 3, and now beta 4, using microbit v2.

Same issue with different Python scripts.

Same issue with different microbit controllers.

Works a few times, then fails with a 521 on the microbit. Disconnecting and reconnecting the microbit cures the problem, but it the recurs. Very reliably fails within a few attempts of flashing, so probably already reported somewhere, but I could not find it.

Using macOS and Big Sur 11.3.

carlosperate, 2021-05-05:
Hi @Maxelon, thanks a lot for the report!

Is it always an error 521? This error would indicate an issue with the hex generation. Generally the final hex created should be deterministic based on the code, so the same Python code using the same version of Mu should always generate the same hex.

If you can replicate this problem with a Python script it would be really useful if you could:
- Share the Python script with us
- Share the hex file Mu creates as well
    - To get a hex file from Mu you can disconnect all micro:bits from the computer and click the "Flash" button. This should ask you were you want to save the hex file, which will add it to a directory on your computer and you could attach to this GH issue as a .txt file or zipped.

One of the reasons sometimes it might or might not fail is that if there is a working MicroPython programme running in the micro:bit, then Mu will first try to send the code directly via serial, instead of generating a hex file.
This serial method is quicker, which is why is the default flashing mechanism, but when it encounters any issues it will fall back to generating the hex and flashing that. At this point if the hex was not generated correctly it will throw that 521 error on the display.

Maxelon, 2021-05-05:
No, it is not always 521, but that seems the usual one for me (but see below).

I've (hopefully) added the code and the hex. Not too many lines, as it was extracted while a play with some ideas. Basically it is a wait loop that monitors the accelerometer, looking for differences. Worked the first time. Failed (with 521) the second.
[wait.py.txt](https://github.com/mu-editor/mu/files/6427313/wait.py.txt)
[micropython.hex.txt](https://github.com/mu-editor/mu/files/6427318/micropython.hex.txt)

I did try repeating this with a simple 'Hello World', but did not get quite the same results. First time through it worked, the second time it gave a 537. If I tried to re-flash, it then worked. Repeated several times and reliably it works, then it fails with 537, then it works again, and so on.

I repeated this experiment with wait.py, and this is reliably working, then failing with 521, then working. I am no longer disconnecting and it seems to work on re-flashing. I thought it was more erratic, but then I was using beta 3, so it could have been different - or maybe I just did not notice. Either way, the pattern seems to be reliable now, flashing successfully every other time. The code always seems to be 537 with 'Hello World' and 521 with my 'Wait.py'.

Maxelon, 2021-05-05:
Just for reference, I have tried creating the hex file from mu several times in succession and then comparing them: The hex file always comes out the same. Probably to be expected, but I thought it helpful to eliminate this as a possible cause.

carlosperate, 2021-05-06:
Thanks @Maxelon! I've been able to replicate the issue with that hex file, which is a great starting point. I'll take a look at this once I find a bit time.