mu-editor/mu #1539: Crash Report

diegolacomba, 2021-05-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: es_ES
Platform: Linux 5.8.0-50-generic #56~20.04.1-Ubuntu SMP Mon Apr 12 21:46:35 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 161, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1

```

carlosperate, 2021-05-06:
Thanks for the report @diegolacomba!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

diegolacomba, 2021-05-06:
[mu.log](https://github.com/mu-editor/mu/files/6434016/mu.log)
 Thanks, waiting for your reply!

carlosperate, 2021-05-06:
Main error seems to be a permissions issue:
```
2021-05-06 12:52:50,760 - mu.virtual_environment:129(download) DEBUG: Collecting pgzero>=1.2.1
  Using cached pgzero-1.2.1-py3-none-any.whl (71 kB)
ERROR: Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 186, in _main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/download.py", line 135, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 177, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 333, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3/dist-packages/pip/_internal/legacy_resolve.py", line 282, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 525, in prepare_linked_requirement
    _copy_file(local_path, download_dir, link)
  File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 133, in _copy_file
    shutil.copy(filename, download_location)
  File "/usr/lib/python3.8/shutil.py", line 415, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.8/shutil.py", line 261, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permiso denegado: '/usr/local/lib/python3.8/dist-packages/mu/wheels/pgzero-1.2.1-py3-none-any.whl'
2021-05-06 12:52:50,760 - mu.wheels:75(remove_dist_files) INFO: Removing wheel/sdist files from /usr/local/lib/python3.8/dist-packages/mu/wheels
2021-05-06 12:52:50,760 - mu.virtual_environment:468(quarantine_failed_venv) INFO: Quarantined /home/diego/.local/share/mu/mu_venv-38-20210506-125248 as /home/diego/.local/share/mu/mu_venv-38-20210506-125248.FAILED
2021-05-06 12:52:50,760 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x7f123a9860d0>; Splash Handler; <SplashLogHandler (INFO)>
2021-05-06 12:52:57,769 - root:173(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 161, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1
```

What commands exactly did you use to install Mu? Did you have to use sudo?

diegolacomba, 2021-05-06:
I only used "sudo pip3 install mu-editor==1.1.0b4"

carlosperate, 2021-05-06:
In that case you might need to run Mu with sudo as well: `sudo mu-editor` 
Does that work?

diegolacomba, 2021-05-06:
Yes it works!! Thanks a lot, have a nice day :)

carlosperate, 2021-05-06:
Great, glad you got it working!

closed 2021-05-06 by carlosperate