mu-editor/mu #1540: Crash Report

brunel-bit, 2021-05-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: fr_FR
Platform: Windows 10 10.0.18362 AMD64

```
Traceback (most recent call last):
  File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\BRUNEL FOTSO\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

carlosperate, 2021-05-06:
Thanks for the report @brunel-bit!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs`

Once we have the log we should merge this with https://github.com/mu-editor/mu/issues/1532.

carlosperate, 2021-06-04:
Hi @brunel-bit, just a friendly reminder in case you'd be able to attach the logs? They would be really helpful for us, and you should be able to find them in `C:\Users\BRUNEL FOTSO\AppData\Local\python\mu\logs`.

carlosperate, 2021-08-06:
Hi @brunel-bit, as this issue is covered in https://github.com/mu-editor/mu/issues/1532 I'm merging this issue over there: https://github.com/mu-editor/mu/issues/1532#issuecomment-894177229

closed 2021-08-06 by carlosperate